danielweidman/pixmob-ir-reverse-engineering #1: Pixmob wristband from Imagine Dragons' Mercury Concert - Toronto - 2022

atulkhatri, 2022-08-23:
Just returned from Imagine Dragons' Mercury concert in Toronto where they distributed PixMob bands.
All the bands had flashing red light at the end. Seems like some kind of 'nuke' command at the end of the concert to let the batteries die.

I ripped open one of the band and the other one is still active after 4 hours at the time of writing.

I really want to make these work but based on my testing I cannot figure out the correct frequency.

I own a mobile with infrared capability and downloaded an app to test various kinds of frequencies but after testing with around 50 remotes, none of them work.

Not sure if the frequencies you posted in the repo would work for this band. Will invest some more time on it. 

Kudos to your work so far. I think you're in the right direction. Wish we hack this soon.

Attaching video & pictures for my band for reference. Feel free to ask any questions.

https://user-images.githubusercontent.com/3920336/186087049-c6e37504-ac7f-4835-8f6f-5560f8a8ecf2.MOV
[image]
[image]

danielweidman, 2022-08-23:
Thanks for the info and teardown photos! I hope the Imagine Dragons concert was good.

I do think the IR codes here would probably work. There's one specific one that might possibly stop the blinking pattern, but it might not even be necessary. I have a bracelet that came from a different concert which can be made to react to signals but only if a bunch are sent right after the batteries are re-inserted, otherwise it reverts to a blinking pattern (which it does go back to eventually after some inactivity).

Of course, in order to test this we need a way for you to be able to transmit the appropriate codes from a device you own (it's unlikely one of the remotes already in the app will work). What kinds of apps are you able to use with your mobile? I wonder if any allow you to specify totally raw IR data (timings and durations of pulses).

If you are willing to invest a little bit, a device like the Broadlink RM mini 3 (or other versions) costs ~$20 USD and can definitely transmit these signals properly. An Arduino device could do it too for a bit cheaper but it would take a few minutes' assembly.

binux14, 2022-08-28:
Hi all!
I was also at the Imagine Dragons' concert in Toronto and I'm testing your sketch right now! 
I was lucky enough that I left just before the command was sent so mine and my wife's bracelet stayed off.

@danielweidman the problem I'm having now is that the Arduino (Arduino MEGA) crashes after I send the serial command via Python. It receives it fine (the RX light flashes) but immediately after the on-board Arduino LED flashes twice indicating it was reset. Any thoughts on this? I know there are some issues with the IRremote library and Arduino MEGA but not sure if that has been solved.

danielweidman, 2022-08-28:
@binux14 Thanks very much for trying this out!

I don't know which pin to use for the Mega, I'm sorry. I've only tested on the Nano and an ESP32 board. I read it might be pin 9 somewhere. Could that be it?

Also, can you try using the serial monitor in the Arduino IDE to send this command to the Arduino?
`[35]19121211241212132213221324121123241,`
That is one of the types of strings the Python code tries to send to the Arduino. If it works via the serial monitor then that would suggest the issue is with how the command is being sent rather than the transmission part. I guess it sounds like that's probably not the case though given what you have observed with the RX light.

danielweidman, 2022-08-28:
If that doesn't work, I could try putting together a sketch that hardcodes just one IR command to send that we could use as a test.

binux14, 2022-08-29:
@danielweidman I got it to work! The command you sent flashes white for 2-3 seconds and fades out.
I ended up using pin 13 (so I have a visual feedback from the on-board LED on the Arduino).
Thank you!

danielweidman, 2022-08-29:
@binux14 Great! To confirm, you can now send commands via the Python code too?

binux14, 2022-08-29:
@danielweidman Yes! Everything is working fine now!
 I seem to have some consistency/power/directionality issue with my single IR LED but that's another thing to troubleshoot on my end.

danielweidman, 2022-08-29:
@binux14 Great!

closed 2022-12-16 by danielweidman

sean1983, 2022-12-16:
Hi All,  Just to clear things up,   Pixmob don't use any Kill or Nuke commands,   Its a bit of a rumor that has been going around for years.   Coldplay's 2nd Gen Xylobands that worked on RF signals rather than Infrared DID include kill codes, and this is where the rumour has stemed from,    This was done thought because their original Xylobands 1st gen had an issue were they would randomly re-trigger due to RF Interference from cellular networks and phones,  The Xylobands used 869.5Mhz band in most countrys which its neighbours bands 896-901 MHz used and still are used by phone networks across the world for GSM, 3G, 4G and now even 5G.

Some people where having to destroy there bands to remove battery's and they where being woken up by them flashing in the night, so people were actually freaked out by it too.

I tried to get mine to do it back then,   but later realized my phone network doesnt use the 900Mhz bands so i was pretty disappointed.

But Pixmob are designed the be reusable / recyclable,  with the ability to easy change the batteries and even been made from more easily recycled materials.