danielweidman/pixmob-ir-reverse-engineering #2: Make bracelets hold colour for 60 seconds

alexmoen955, 2022-08-24:
First of all, wonderful project that allows you to relive a concert. You literally get chills again :)

I'm experimenting a lot with the effects and tail codes but I was wondering if it's possible to make a colour last for 60 seconds?
What do you mean by "Transmitting "STP" right after a X_THEN_Y"?

(this isn't really an issue but I didn't find another way to ask the question)

Greetings
Alex

danielweidman, 2022-08-24:
Hi Alex,

Thanks so much for looking at the notes that carefully! Sorry that part wasn't clear at all, it lines up with old names of signals.

Can you let me know what version of bracelet you have? I.e. when did you get it/what does it look like? (a photo would be ideal).

As far as I've found, the 60-second color hold thing only works on certain older versions of the bracelet. I can make better instructions though and keep trying different command orders to see if I can make it work on newer wristbands.

-Dani

alexmoen955, 2022-08-24:
My bracelet is from the Music Of The Spheres tour.

[image]
.

danielweidman, 2022-08-24:
Got it. I haven't seen a color-hold effect for that yet, but I'm trying some stuff now.

To be clear, you are looking for it to hold the color after no longer being in IR range, right? It can definitely be made to hold it's color while in IR range by just resending one of the color commands (with no fade effect) repeatedly.

alexmoen955, 2022-08-24:
Oh cool! Is there a way to repeat a command x times (without rewriting the command multiple times)?

EFFECTS_TO_SHOW = [
    {
        "main_effect": "RED",
        "tail_code": None,
        "duration": 0
    },
    {
        "main_effect": "RED",
        "tail_code": None,
        "duration": 0
    },
    {
        "main_effect": "RED",
        "tail_code": None,
        "duration": 0
    },
    {
        "main_effect": "RED",
        "tail_code": None,
        "duration": 0
    },
    {
        "main_effect": "RED",
        "tail_code": None,
        "duration": 0
    },
    {
        "main_effect": "RED",
        "tail_code": None,
        "duration": 0
    },
    {
        "main_effect": "RED",
        "tail_code": None,
        "duration": 0
    },

danielweidman, 2022-08-24:
Good point!

Let me make a quick edit to the `demo_multiple_effects.py` script to add that feature.

danielweidman, 2022-08-24:
@alexmoen955 Okay! I made a new file instead of editing `demo_multiple_effects.py`

Try running `demo_multiple_effects_advanced.py`. Now, you can optionally add `"hold_with_repeated_send": True` to your entries which will make them hold the color for as long as is specified in the `duration`.

It only works as expected for simple color entries (from `base_color_effects`) with no tail code.

alexmoen955, 2022-08-24:
It looks like it only works when the bracelets are very close to the IR sensor. When the bracelets are located a bit further they start to strobe.

danielweidman, 2022-08-24:
Ah yeah, that can be an issue. As yeah, that's gonna be an issue if the IR reception isn't great. If it doesn't receive some of the re-transmissions then it may not keep the light on steadily.

Replacing the batteries can make a big difference in improving the reception, as will going to a dark room, and/or getting a more powerful transmitter. Also, some of the `base_color_effects` have longer base durations than others. For example, `GREEN_11` lasts longer than `GREEN` and might make less strobing.

jorgelml1, 2022-09-18:
guys can u help me? Im trying to get my  music of the spheres xyloband back to live. Im using an esp8266 and a LED IR.

I ran the code but dont know what to expect.. Should I write something else in the console window?

danielweidman, 2022-09-18:
> guys can u help me? Im trying to get my music of the spheres xyloband back to live. Im using an esp8266 and a LED IR.
> 
> I ran the code but dont know what to expect.. Should I write something else in the console window?

Hello! Just to confirm, did you both A) load the Arduino code onto the ESP8266 and B) run the Python code with the ESP8266 connected to the computer?

jorgelml1, 2022-09-18:
Hi, Daniel

I've just ran the demo_single_effect python code. Nothing happens, LED IR emitter works fine. I did a blynk test and everything seems to be ok.

danielweidman, 2022-09-18:
1. What was the output of the Arduino IDE when you loaded on the sketch from the arduino_sender folder?
2. Did you set the COM port correctly in the python code?
3. Have you tried setting WAIT_BEFORE_SEND to True in the python code?

jorgelml1, 2022-09-18:
Seems that the arduino code its not properly working since i used my camera to see if there was some lighting from IR led. 

What version of IDE arduino are u using? 
My board is NODEMCU V3

danielweidman, 2022-09-18:
I don't know exactly what version of the Arduino IDE I used and I don't have it in front of me (I can check later).
Also:

1. Which of the Arduino scripts from the arduino_sender folder did you use?
2. What was the output of the Arduino IDE when you loaded on the sketch from the arduino_sender folder?
3. Did you set the COM port correctly in the python code?
4. Have you tried setting WAIT_BEFORE_SEND to True in the python code?

jorgelml1, 2022-09-18:
Daniel,

1. I use the PixMob_Transmitter_ESP32
2. Here is what i got
Executable segment sizes:

ICACHE : 32768           - flash instruction cache 

IROM   : 237460          - code in flash         (default or ICACHE_FLASH_ATTR) 

IRAM   : 27289   / 32768 - code in IRAM          (IRAM_ATTR, ISRs...) 

DATA   : 1496  )         - initialized variables (global, static) in RAM/HEAP 

RODATA : 888   ) / 81920 - constants             (global, static) in RAM/HEAP 

BSS    : 25712 )         - zeroed variables      (global, static) in RAM/HEAP 

El Sketch usa 267.133 bytes (25%) del espacio de almacenamiento de programa. El máximo es 1.044.464 bytes.
Las variables Globales usan 28.096 bytes (34%) de la memoria dinámica, dejando 53.824 bytes para las variables locales. El máximo es 81.920 bytes.
esptool.py v3.0
Serial port COM7
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: c4:5b:be:5d:22:84
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 271280 bytes to 199011...
Writing at 0x00000000... (7 %)
Writing at 0x00004000... (15 %)
Writing at 0x00008000... (23 %)
Writing at 0x0000c000... (30 %)
Writing at 0x00010000... (38 %)
Writing at 0x00014000... (46 %)
Writing at 0x00018000... (53 %)
Writing at 0x0001c000... (61 %)
Writing at 0x00020000... (69 %)
Writing at 0x00024000... (76 %)
Writing at 0x00028000... (84 %)
Writing at 0x0002c000... (92 %)
Writing at 0x00030000... (100 %)
Wrote 271280 bytes (199011 compressed) at 0x00000000 in 17.6 seconds (effective 123.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

3. COMs are the same COM7 in my case
4. Yep, Ive changed to True but its not working.

Ive deleted the BT code in the IDE arduino, btw.

danielweidman, 2022-09-18:
Cool, good thinking to delete the BT code. Hmmm.

1. What happens if you open up the Serial Monitor and paste in this command: `[35]19121211241212132213221324121123241,`
2. Did you configure to IR LED pin number in the script to match the one it is connected to? Sometimes the numbers are confusing and different between what is written on the board and what is supposed to be set on the IDE.

jorgelml1, 2022-09-19:
1. I pasted the command but nothing happens. Ive just blynk again pin4 in my board and its ok.
2. Yep, is in D2 which is supposed to be GPIO 4

danielweidman, 2022-09-19:
Thanks for all the info and checks.

Have you changed the batteries in your bracelet since the show? I'd expect them to last until now assuming you went to a show in Perú but maybe not. I know you said you checked if the LED was working with the camera, but since it would only be on for a few milleseconds I wonder if the camera is missing that.

Otherwise, I'll have to keep thinking.

jorgelml1, 2022-09-19:
Batteries are ok. Blynking should be barely showed in demo_multiple_effects but the IR is still off. Ill be think about it too. Ill try another board, nano could be the reasonable option.

danielweidman, 2022-09-19:
Good luck!

closed 2022-12-16 by danielweidman