danielweidman/pixmob-ir-reverse-engineering #3: Circuit

jacojaco12, 2022-09-22:
Hey Daniel! I just have a basic question here... Could you please send me an image of your circuit? The one you suggested hasn't quite worked for me :/ Thank you :)

danielweidman, 2022-09-22:
Hi there! I can send an image later. For now though, can you let me know if you are using a premade IR module (like this: https://www.amazon.com/Dorhea-Transmitter-Sensor-Electronic-Building/dp/B08X2P7P11/) or a raw IR led and resistor and/or transistor?

jacojaco12, 2022-09-23:
Hey! Thanks for being so attentive :) I'm indeed using a premade IR module. As for the microcontroller, im using an ESP-WROOM-32.

danielweidman, 2022-09-23:
Cool! (and no problem).

Here is basically what I have for my ESP32:

[image]

Does that match yours?

IIUC, it's maybe not considered best practice to use VIN for the 5V supply as I do, but since it's connected by USB that pin will provide 5V out. You could alternatively connect the VCC/5V pin on the IR module to the 3V3 pin on the ESP32 board, but it might make the IR signal weaker.

jacojaco12, 2022-09-24:
Jesus, it finally turned on. However, both Pixmobs seem unresponsive. I'm certain it's not their batteries, as I opened one of them just to check if they were still alive, which they in fact are. It isn't the IR Led either, as I can see it turned on with my VR. Any ideas on what might be causing this?

danielweidman, 2022-10-11:
What if you try removing and reinserting the batteries while sending IR commands quickly?

Vetmom79, 2023-09-05:
Hi Daniel! I am using an Arduino Uno and a premade IR module for my pixmob bracelet. Can you send me a quick schematic for your circuit? This is my first project EVER because my bracelet didn't receive the stay on code at the end of the Eras Tour and I desperately just want it to turn on (nothing fancy). I am a 43 year old veterinarian and mom and am at bare basics. 🥴

danielweidman, 2023-09-09:
> Hi Daniel! I am using an Arduino Uno and a premade IR module for my pixmob bracelet. Can you send me a quick schematic for your circuit? This is my first project EVER because my bracelet didn't [...]

@Vetmom79 and I figured this out on Discord! The issue was that sometimes WAIT_BEFORE_SEND needs to be set to True in config.py. I've just updated it to be True by default.

xristos-sk, 2025-03-23:
what is Discord ? 

Here is a very good schematic and explanation for anyone needs to see the answer on the creation of the LED module:

https://github.com/peterhinch/micropython_ir/blob/master/TRANSMITTER.md