danielweidman/pixmob-ir-reverse-engineering #4: Bad Bunny Pixmob Bracelet

davidmendoza628, 2022-10-11:
Hi,

I built your suggested IR circuit and I'm trying to use your Arduino/Python code but something is not working. 

My bracelet seems to be stuck in a mode that just fades in and out of different colors. Is it possible that they nuked my bracelet at the end of the show? Do you know how to put the bracelet back into a mode that accepts commands?

Thank you,
David Mendoza

danielweidman, 2022-10-11:
Hi there,

It's possible, though I've had some luck by taking the battery out, putting it back in, and immediately sending lots of IR commands quickly (maybe in a for loop with a short delay).

If that doesn't work with normal color commands, maybe try some of the special commands starting with "WEIRD_", like the ones in the first block of effect_definitions.py under "# Weird stuff (multiple random colors, set impact future effects, etc):"

juanbracamonte, 2022-11-13:
@davidmendoza628 Yo pude controlar las bandas de badb. sds

closed 2022-12-16 by danielweidman