danielweidman/pixmob-ir-reverse-engineering #5: Message protocol

JSMSolns, 2022-10-18:
Firstly - brilliant work on decoding the messages to the wrist bands.
I've written an Arduino script to take your bit sequences and turn them into IR timings for the transmitter and it all works a treat on 2 Music of the Spheres wristbands.

I've been struggling with trying to understand the message formats though? 
I was hoping to control these via some DMX software (VixenLights) sync'd to an audio track and using the Arduino to decode the DMX and convert to suitable messages. I was expecting some sort of RGB(W) levels to be embedded in the messages but that doesn't seem to be the case here?

It would be interesting to know how the professional lighting controllers sequence these wristbands.

One thing I've noticed (as I'm sure you have too) is that there is never any more than 2 '1' bits together which makes me think there is some sort of additional encoding going on? Clearly not Manchester coding as there are too many repeating 0s.

danielweidman, 2022-10-20:
Thanks for the kind words! Glad to hear it is working for you.

When I started this project, I was also expecting some sort of clear RGB levels encoded in the messages, but so far I have not been able to figure out the exact pattern. There's got to be one though, unless the bracelets literally store a table of IR code to light effect/color. 

I've found that usually if you discover a working IR code via brute force that's just a couple bits off from another, the color is usually similar (like light green vs dark green). However, when I (and my partner) tried segmenting out the packets in various ways based on estimates of what we thought might be the R, G, and B (or H, S, and V)-associated bits and then changed one to try and get a desired color, most of the time the resulting code wouldn't work (leading to a theory that there's a checksum of some sort too). 

It's of course notable that each packet, in addition to just the color, also seems to indicate the duration of the effect (and sometimes the fade parameters). So there's some information in each beyond just the RGB levels; that's made it harder to identify the pattern.

FWIW, I came across this page during my search, maybe it could reveal something? 
https://onyxfixturefinder.com/fixture/PixMob/pixmob+pro2.1
Would be interested to hear your thoughts on that if you understand it better than me. I believe it mainly has to do with how the controller/computer/whatever communicates with the PixMob IR blaster (which takes commands via DMX) though. So the PixMob transmitter is doing some extra translation of the signals from DMX to get the output IR commands. That's my understanding at least.

Here's the manual for one model of PixMob transmitter that takes commands according to the "PixMob PRO 2" protocol: https://fccid.io/2ADS4WASH/User-Manual/Users-Manual-4434103.pdf

Anyway, those are just some ramblings; it would be great to fully figure out the pattern.

JSMSolns, 2022-10-21:
Hi Dani,Thanks for the reply and the links. The first link must be the DMX channel allocations and there are a lot more of them than I thought there would be. The second link I'd already come across and is useful for the overall understanding but doesn't really give any more technical details.I still haven't figured out the coding and like you, I tried to examine similar messages (green, light green etc.)to see which bits changed but haven't seen a pattern yet.Initially, I too felt there must be a checksum but looking at very similar messages that doesn't seem to be the case, particularly if the checksum is at the end of the message which is the normal position.If my theory about only seeing a maximum of 2 consecutive 1s anywhere in a message is true, the number of permutations is considerably reduced. I would also guess from your work that the first bit is always a '1' and the last 2 bits are always '01' which would give a 36 bit basic message?For now I've given up on trying to work out the encoding and have resorted to just picking out a set of basic colours from your lists. I've got about 17 at the moment (I've ignored packets which give very similar colours) In the Vixen software, I'll set up some 'favourite' colours to match these 17 wristband colours so I can just drop them onto the timeline. The DMX is sent over serial to the Arduino which will take the RGB and find the closest match (i.e. which of the 17 is nearest). It then transmits the appropriate IR sequence.  The Arduino will automatically add any additional STP commands if required for consecutive commands (not yet sure what the minimum delay is between commands before a STP is needed?).If/when I get that working, I'll add another DMX channel purely for the effects (fade in, fade out, random etc.). The Arduino will read those and transmit the appropriate sequence at the end of the colour transmission.Much more work than just sending the RGB levels but should be reasonably effective in the end I hope.If I get it working, I'll send you a sample video.Keep up the great work.John

https://onyxfixturefinder.com/fixture/PixMob/pixmob+pro2.1

danielweidman, 2022-10-31:
Hi again,

That's really cool what you've been doing with the DMX signal and the selection of the closes color by RGB values! I love it and would be very interested in seeing a sample video.

Regarding the minimum acceptable delay, I can tell you that the recordings we've gotten indicate that PixMob leaves 6300 microseconds (equivalent to 9 700 microsecond "0" bits) between separate messages/repetitions of messages.

JSMSolns, 2022-11-01:
Hi,
I managed to make a bit of progress.
Here is a link[link](https://drive.google.com/file/d/1RbWuvI1nAY7a7wVZ3fR-A8afUV0r4NzG/view?usp=drivesdk)  hopefully to a video I made of my set up running with 2 wristbands. I covered the wristbands with a sheet of paper because otherwise their brightness saturates the camera and makes it hard to discern the colours.
The setup is running Vixen with the DMX output sent over a serial (usb) port. The Arduino reads the serial DMX stream, extracts the channel information, interprets it and re-encodes it as an IR stream and transmits it to the IR sender attached to the Arduino.
The channels set up are:
1: Red
2: Green
3: Blue
4: Tail code index
5: Weird code index
Hope you like Coldplay! 
I'll leave the link open for a few days. 
Let me know if you manage this view it ok.

danielweidman, 2022-11-01:
Brilliant! You've got a whole thing set up end-to-end and it look so cool. Thanks very much for sharing the video.

Are there any particular shades of color/effects/etc you feel are missing? I got some new recordings the other day (will add to this repo) and am hoping to learn a bit more about the packet structure.

sean1983, 2022-11-01:
> ID: <danielweidman/pixmob-ir-reverse-engineering/issues/5/1299292586@
> github.com>

Bloody hell mate,  you've done it.  and it looks fantastic!

This is exactly what I have been trying too do with no luck.

Vixen is just the original version of xLights isn't it?

Soon,   I guess this could run on an ESP32 also?

Is your code available?   I really really want to play!

Specially know I have a collection of about 20 of these latest gen bands
from Coldplay's last tour!

I would love to use them with xLights or my DMX Console with my Christmas
Lights Display aswell as for Party's,  I would love to actually sequence
them to a whole video of a concert matching the fx being used.  Better then
VR,   Although I might have to build some CO2 & Concetti Cannons too!

I saw them twice this year,  Wembly and the second time in Glasgow.  And at
Glasgow I had a little play,   I took a Cheap Home Automation IR Hub
(SwitchBot) with me, Programmed up with tons of this IR codes,  Powered it
USB-C on my phone and connected it via WiFi to my phones Hotspot,   Now I
wouldn't spoil a Coldplay Gig,  But before them started I held it in my
hand behind my phone and push a few codes out.  🤣🤣🤣

I got about 12 seats each side of me and 6 or 7 rows up and down.    The
look on people's faces thinking the gig was about to start was priceless.

Didn't want to try holding it up,  because it was probably obvious enough
to the lighting directors,  centre of where they were all flashing away,
Plus this PixMob Moving Heads have IR / Visible Cameras on board for
Calibration at venue's so they definitely would have seen me!

But leaving the venue was fun watching 100s of them flash ad I walked to
find a taxi.

Well done on the progress mate.  👏

Sean Yem

JSMSolns, 2022-11-02:
> Brilliant! You've got a whole thing set up end-to-end and it look so cool. Thanks very much for sharing the video.
> 
> Are there any particular shades of color/effects/etc you feel are missing? I got some new recordings the other day (will add to this repo) and am hoping to figure out a bit more about the packet structure.

Cheers for that.
Just wanted to see what was possible rather than having a particular use case. Wouldn't have been possible without all your hard work, so thanks again for doing it all.

Most of the video was done just using the RGB settings with just a few of the tail codes and weird codes towards the end (slow fades etc.) as these were a late addition to the development.
I'll probably park the development for now, so don't worry about additional decodes on my part - plenty to go at already!
On the protocol front, I have a suspicion that the coding may be based on 3 bit encoding but excluding 111 ( ignoring the first and last bits as start/stop)  but this still doesn't get me any further.

JSMSolns, 2022-11-02:
> Bloody hell mate, you've done it. and it looks fantastic! This is exactly what I have been trying too do with no luck. Vixen is just the original version of xLights isn't it? Soon, I guess this [...]

Thanks for the comment.
My intention was always to follow in the footsteps of Coldplay in terms of reuse/recycling/sustainability and try and use these wrist bands rather than throw them away. Possibly to use them at parties/Xmas.

Arduino code isn't complicated. Just loops reading the serial port and picking out the channel info (mainly RGB). As there doesn't seem to be a direct coding of colour on the IR message, the Arduino tries to do a closest match to the known colours available (these are also used as favourites in the colour selection within Vixen). The ir code sequences are held in 2d arrays in the program memory as 0s and 1s as shown on this site. The code then converts them to 700us timings in real time to pump out to the IR led.
I think you can also use Xlights in a similar way. You can also use an ESP32 but may have to ensure interrupts are disabled whilst transmitting the IR possibly?

sean1983, 2022-11-05:
Its impressive,   I am.not a coder,    but can normally wangle something together copy and pasting bits here and these and generally get something to working.

But no look with this,  I need to learn more about programming arduino,  Brain is getting older and visually impaired so things are much more difficult these days.

I already Daniels code on the ESP32 but rather than using a single IR LED I have the data pin drive a Transistor that is firing off an Intrared Illuminator PCB used for CCTV,  so can trigger the bands over a large area now!  

I am trying to add RS485 to I can have a standard DMX Input,  plus also got an Ethernet module that will work with the ESP32 but take up far too many pins,  but that was a thought of running hard wired Ethernet for ArtNet or E131.  Obviously them protocols can run over WiFi but latency is an issue then.  Direct RS458 (DMX) should be easier to implement at least for me.

Would you be able to share the code you have written so far to make this work with Vixen please 🙏 it would really  help me,  I understand if you don't.

Thanks in advance.

Sean

JSMSolns, 2022-11-05:
Happy to share the code although it is very much a development setup and could be tidied up a lot. You will probably have to modify it for the ESP32 and the RS485 handling. 
I'd recommend setting it up for OTA updates because having to constantly switch the serial port between programming and Vixen was a bit of a pain.
I'll try and upload the code later tonight or tomorrow and send a link.
I hope the basis of this code won't be used at any live concerts and only used to re-use the wristbands for personal use.

sean1983, 2022-11-05:
Thanks mate,

I guess Vixen just seeing the Arduino as a USB DMX controller ?

Good call on OTA, If I can get the RS-485 working then i can just use one
of my OpenDMX or uDMX dongles from Vixen/xLights and leave the onboard
serial for Programming and Monitoring.

Don't worry code will only be used for personal use.

As I mentioned,  I'm visually impaired (Registed Blind) and don't work,   I
would have loved to go into the lighting industry,  I did study Sound &
Light tech in collage,  but working out on Lighting grids or dark venues
wouldnt have been possible for me, I just fascination with Lighting
Fixtures, Moving Heads, Anything LED, Addresable Pixels and DMX and ArtNet
and other systems,

So code is just to re-use my collection of Wristbands,  And being a massive
coldplay fan I am of the same ethos of recycle, re-use etc.  One thing to
note,  I wish i could get Rechargeable CR-2032's. Maybe i need to LiPo mod
them.

I also have one early Pixel Mob bands I have not even looked at yet,  They
are from a Spice Girls concert in Manchester 2019,  they are alot smaller,
probably only the motion activated ones,  Most open them.

https://drive.google.com/file/d/1QE-8lpl12X_XBjhL1uX-io3jyrxlxwb5/view

I also have a few of the Original single cold RF Xylobands, and some of the
RGB ones that have a speaker inside,  I believe there is a project that has
these working too.

https://drive.google.com/file/d/1QDiB-I6Yjiro69RZL-q77J8kvs3rYY7Q/view

Anyways sorry for the rambling on,

If you feel more comfortable your welcome to just share it via Google Drive:

 ***@***.***

Cheers,

Sean Yem

JSMSolns, 2022-11-05:
The Arduino Nano development code is here:

https://drive.google.com/drive/folders/1QL1kPWrWzF3vtw2RiBbOQsQR6VMyGKgv?usp=sharing

I'll leave it open for a day or two for you to grab.
There are some comments at the top of the code which may help with setting up depending upon how you're going to interface your ESP32.

The Vixen software is set up as a Generic serial controller with a packet header of +> (so the Arduino knows where the start of a DMX message begins).  The Arduino is plugged into the PC's USB port and receives the serial output from Vixen.
I guess your openDMX/RS-485 interface won't require this.

Hope you can make sense of my coding (containing very few comments!). There is quite a bit of copy/paste of code due to the development phase so could be tidied up a lot by using more common functions.

If you struggle to understand any of the code, let me know and I'll try and help out where I can.

sean1983, 2022-11-06:
Hi Mate,  All working nicely on the ESP32 after hours of tinkering,  Damn
IR Library is different, Thanks to Daniels ESP code I was able to resolve
it,  I sent longer email and link to video to the email you shared code
from,  I'm just out buying AAA's now!

JSMSolns, 2022-11-06:
Glad you got it working on the ESP32.
Enjoy.

closed 2022-11-06 by JSMSolns