danielweidman/pixmob-ir-reverse-engineering #6: Compile Errors

AddumBacon, 2022-10-24:
I'm new to Arduino and compiling code, but want to mess with the Pixmob light I got.

I tried to compile/upload the Transmitter sketch and got these errors:

C:\Users\Adam\Desktop\PixMob_Transmitter_Arduino\PixMob_Transmitter_Arduino.ino:1099:25: error: missing terminating " character
             data-action="
                         ^
C:\Users\Adam\Desktop\PixMob_Transmitter_Arduino\PixMob_Transmitter_Arduino.ino:1100:41: error: stray '#' in program
               input-entered:ref-selector#inputEntered
                                         ^
C:\Users\Adam\Desktop\PixMob_Transmitter_Arduino\PixMob_Transmitter_Arduino.ino:1101:40: error: stray '#' in program
               tab-selected:ref-selector#tabSelected
                                        ^
C:\Users\Adam\Desktop\PixMob_Transmitter_Arduino\PixMob_Transmitter_Arduino.ino:1102:38: error: stray '#' in program
               focus-list:ref-selector#focusFirstListMember
                                      ^
C:\Users\Adam\Desktop\PixMob_Transmitter_Arduino\PixMob_Transmitter_Arduino.ino:1103:13: error: missing terminating " character
             "
             ^
C:\Users\Adam\Desktop\PixMob_Transmitter_Arduino\PixMob_Transmitter_Arduino.ino:1150:25: error: missing terminating " character
             data-action="
                         ^
C:\Users\Adam\Desktop\PixMob_Transmitter_Arduino\PixMob_Transmitter_Arduino.ino:1151:41: error: stray '#' in program
               input-entered:ref-selector#inputEntered
                                         ^
C:\Users\Adam\Desktop\PixMob_Transmitter_Arduino\PixMob_Transmitter_Arduino.ino:1152:40: error: stray '#' in program
               tab-selected:ref-selector#tabSelected
                                        ^
C:\Users\Adam\Desktop\PixMob_Transmitter_Arduino\PixMob_Transmitter_Arduino.ino:1153:38: error: stray '#' in program
               focus-list:ref-selector#focusFirstListMember
                                      ^
C:\Users\Adam\Desktop\PixMob_Transmitter_Arduino\PixMob_Transmitter_Arduino.ino:1154:13: error: missing terminating " character
             "
             ^
C:\Users\Adam\Desktop\PixMob_Transmitter_Arduino\PixMob_Transmitter_Arduino.ino:1637:72: error: stray '#' in program
           <td id="LC35" class="blob-code blob-code-inner js-file-line">#<span class="pl-k">include</span> <span class="pl-s"><span class="pl-pds">&lt;</span>Arduino.h<span class="pl-pds"></span></span></td>
                                                                        ^
C:\Users\Adam\Desktop\PixMob_Transmitter_Arduino\PixMob_Transmitter_Arduino.ino:1641:72: error: stray '#' in program
           <td id="LC36" class="blob-code blob-code-inner js-file-line">#<span class="pl-k">include</span> <span class="pl-s"><span class="pl-pds">&lt;</span>IRremote.hpp<span class="pl-pds"></span></span></td>
                                                                        ^
C:\Users\Adam\Desktop\PixMob_Transmitter_Arduino\PixMob_Transmitter_Arduino.ino:1704:182: error: stray '#' in program
           <td id="LC51" class="blob-code blob-code-inner js-file-line">  incomingString = Serial.<span class="pl-c1">readStringUntil</span>(<span class="pl-s"><span class="pl-pds">&#39;</span>[<span class="pl-pds">&#39;</span></span>); <span class="pl-c"><span class="pl-c">//</span> read the incoming byte:</span></td>
                                                                                                                                                                                      ^
C:\Users\Adam\Desktop\PixMob_Transmitter_Arduino\PixMob_Transmitter_Arduino.ino:1704:216: error: stray '#' in program
           <td id="LC51" class="blob-code blob-code-inner js-file-line">  incomingString = Serial.<span class="pl-c1">readStringUntil</span>(<span class="pl-s"><span class="pl-pds">&#39;</span>[<span class="pl-pds">&#39;</span></span>); <span class="pl-c"><span class="pl-c">//</span> read the incoming byte:</span></td>
                                                                                                                                                                                                                        ^
C:\Users\Adam\Desktop\PixMob_Transmitter_Arduino\PixMob_Transmitter_Arduino.ino:1708:182: error: stray '#' in program
           <td id="LC52" class="blob-code blob-code-inner js-file-line">  incomingString = Serial.<span class="pl-c1">readStringUntil</span>(<span class="pl-s"><span class="pl-pds">&#39;</span>]<span class="pl-pds">&#39;</span></span>); <span class="pl-c"><span class="pl-c">//</span> read the incoming byte:</span></td>
                                                                                                                                                                                      ^
C:\Users\Adam\Desktop\PixMob_Transmitter_Arduino\PixMob_Transmitter_Arduino.ino:1708:216: error: stray '#' in program
           <td id="LC52" class="blob-code blob-code-inner js-file-line">  incomingString = Serial.<span class="pl-c1">readStringUntil</span>(<span class="pl-s"><span class="pl-pds">&#39;</span>]<span class="pl-pds">&#39;</span></span>); <span class="pl-c"><span class="pl-c">//</span> read the incoming byte:</span></td>
                                                                                                                                                                                                                        ^
C:\Users\Adam\Desktop\PixMob_Transmitter_Arduino\PixMob_Transmitter_Arduino.ino:1725:182: error: stray '#' in program
           <td id="LC56" class="blob-code blob-code-inner js-file-line">  String newVals = Serial.<span class="pl-c1">readStringUntil</span>(<span class="pl-s"><span class="pl-pds">&#39;</span>,<span class="pl-pds">&#39;</span></span>);</td>
                                                                                                                                                                                      ^
C:\Users\Adam\Desktop\PixMob_Transmitter_Arduino\PixMob_Transmitter_Arduino.ino:1725:216: error: stray '#' in program
           <td id="LC56" class="blob-code blob-code-inner js-file-line">  String newVals = Serial.<span class="pl-c1">readStringUntil</span>(<span class="pl-s"><span class="pl-pds">&#39;</span>,<span class="pl-pds">&#39;</span></span>);</td>
                                                                                                                                                                                                                        ^
C:\Users\Adam\Desktop\PixMob_Transmitter_Arduino\PixMob_Transmitter_Arduino.ino:1795:12: error: missing terminating ' character
       </option></form><form class="js-jump-to-line-form Box-body d-flex" data-turbo="false" action="" accept-charset="UTF-8" method="get">
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\Adam\Desktop\PixMob_Transmitter_Arduino\PixMob_Transmitter_Arduino.ino:1870:12: error: stray '\342' in program
     You can’t perform that action at this time.
            ^
C:\Users\Adam\Desktop\PixMob_Transmitter_Arduino\PixMob_Transmitter_Arduino.ino:1870:13: error: stray '\200' in program
     You can’t perform that action at this time.
             ^
C:\Users\Adam\Desktop\PixMob_Transmitter_Arduino\PixMob_Transmitter_Arduino.ino:1870:14: error: stray '\231' in program
     You can’t perform that action at this time.
              ^
[... 16 lines cut ...]
C:\Users\Adam\Desktop\PixMob_Transmitter_Arduino\PixMob_Transmitter_Arduino.ino:1176:97: error: expected unqualified-id before '<' token
     <span class="flex-1 css-truncate css-truncate-overflow {{ isFilteringClass }}">{{ refName }}</span>
                                                                                                 ^
C:\Users\Adam\Desktop\PixMob_Transmitter_Arduino\PixMob_Transmitter_Arduino.ino:1637:151: error: expected unqualified-id before '<' token
           <td id="LC35" class="blob-code blob-code-inner js-file-line">#<span class="pl-k">include</span> <span class="pl-s"><span class="pl-pds">&lt;</span>Arduino.h<span class="pl-pds"></span></span></td>
                                                                                                                                                       ^
[... cut ...]
C:\Users\Adam\Desktop\PixMob_Transmitter_Arduino\PixMob_Transmitter_Arduino.ino:1663:185: error: expected unqualified-id before '<' token
           <td id="LC41" class="blob-code blob-code-inner js-file-line">String incomingString = <span class="pl-s"><span class="pl-pds">&quot;</span><span class="pl-pds"></span></span>;</td>
                                                                                                                                                                                         ^

exit status 1

Compilation error: missing terminating " character

Any suggestions?
Thanks!

danielweidman, 2022-10-28:
Hi there! Would you be able to provide large, high resolution screenshot(s) showing the entire Arduino IDE window, including the compile errors?

AddumBacon, 2022-10-30:
Actually, I input the code wrong.
The code compiles and uploads, but I don't think the light is working
Here is my Arduino setup
(https://drive.google.com/file/d/1rRBhOav8IWJbG1fFa1jqznPeRbfilMcw/view?usp=sharing)

danielweidman, 2022-10-31:
It looks like line 32 is:
`const uint16_t kIrLed = ~3;`

Can you make it just:
`const uint16_t kIrLed = 3;`

AddumBacon, 2022-11-01:
I changed it, but it still isn't working.
[image]

danielweidman, 2022-11-01:
Got it.

Can you describe exactly what steps you are taking to try and get the bracelet to light up (connecting the Arduino, loading the Arduino code, running the Python code, etc)?

Also, which PixMob bracelet/device do you have and have you replaced the batteries recently?

AddumBacon, 2022-11-01:
Ok, so this is my fault. I haven't installed or worked with python before. I was trying to figure it out, but I'm on Windows and it is a bit confusing.

The PixMob is from The Weeknd show in Seattle on August 25th.

danielweidman, 2022-11-01:
Hi there.

No problem, I'm also on Windows and it can be a little annoying but we should be able to get it figured out. There's one way we can do a quick test before doing stuff with Python.

You'd want to first install the official Arduino IDE, install the required IRemote library, take the code from your screenshot and load it onto the Arduino. I'm not sure exactly which of those steps you've done already. You can probably find tutorials about them but I am happy to try and answer questions too.

Also, you will definitely need to replace the battery for the The Weeknd bracelet if you haven't yet. It takes two CR1632 batteries.

danielweidman, 2022-11-01:
Once you have the code running on the Arduino and you have it connected to the computer via USB, you can test the setup without Python by opening the Serial Monitor window in the Arduino IDE, confirming the baud rate is set correctly to match the code (115200), pasting in the following text (including the trailing comma), and sending it: 
`[35]19121211241212132213221324121123241,`
(sounds like there's still a few more things to be done before that though)

AddumBacon, 2022-11-01:
I did all that and didn't see any info come back on the Serial Monitor

[image]

danielweidman, 2022-11-01:
I see.

Two things:
1. Did you press enter after pasting that text in the Serial Monitor?
2. You shouldn't expect to get anything back--if the circuit is set up right and the bracelet has power it should react.

AddumBacon, 2022-11-01:
Yes, I pressed enter and a light on the board did light up.
Looks like I need to get some new batteries. I will let you know how it goes after I do that!

danielweidman, 2022-11-01:
Good luck!

AddumBacon, 2022-11-02:
Ok, new batteries and it still isn't working.

AddumBacon, 2022-11-03:
Nevermind, the batteries weren't in correctly! One has to be upside down.

It works!

[image]

danielweidman, 2022-11-03:
Cool. Biggest things to note may be:

1. When installing Python, check the both that says "Add Python to PATH" or something similar (if you already installed it and didn't check that box it's possible to do it later but it is just a little harder)
2. After installing it, open a command prompt window in the python_host folder (of this repo) and run the command "pip install -r requirements.txt"
3. Edit the line 23 of  "demo_single_effect.py" to specify the same COM port as you selected in the Arduino IDE to make that work
4. Set the variable on line 29 to True
5. Run the "demo_single_effect.py" file

I also won't be mad if you don't want to bother haha

danielweidman, 2022-11-03:
Great!

AddumBacon, 2022-11-04:
I changed the parameters and ran the code and got back these errors.
[image]

danielweidman, 2022-11-04:
Do you still have the serial monitor open? If so, that may be "taking over" that port and making it so the Python program can't access it.

AddumBacon, 2022-11-04:
Legend! That worked.
Thanks so much for sticking with me and helping me figure it out!

https://photos.app.goo.gl/Zh8rPKEWw74ZWmPo6

AddumBacon, 2022-11-04:
Here are some more pics. I have two of them.

[image][image][image]

danielweidman, 2022-11-04:
Yay!

closed 2022-12-16 by danielweidman