danielweidman/pixmob-ir-reverse-engineering #8: Colour Mixing and Encoding

sean1983, 2022-11-11:
Hi Daniel,

I saw you mention about trying to figure out the Colour mixing,  RGB HSV etc,
Did you try CMY,  The theatrical industry is used to using Cyan Magenta Yellow from the Colourwheel mixing in Moving Heads, or Scrollers on static lights.

I was looking at the BT LE Pixmob project on github and wondered if the encoding might be simlar,  Looking at the DMX Channel Breakdown from the Onyx Lighting Console vs the BT LE Bands,   It kinda looks like PixMob would probably used the same incoding,  or at least very simlar!  e.g. Below.

Cheers,

Byte 0		Always EE
Byte 1		Flags field:
		Bit 1: Clap mode
		Bit 2: Multicolor mode, select random or sequential from a pool of colors.
		Bit 5: One-shot mode, ignore further identical commands.
Byte 2		Green component of color data (00 is off)
Byte 3		Red component of color data (00 is off)
Byte 4		Blue component of color data (00 is off)
Byte 5		Chance and attack
		Bit 0-2:  1: 85% -> 7: 5%
		Bit 3-5: attack:  0: instant-> 7: slowest
		Bit 6-7: ignored
Byte 6:		Sustain and release
		Bit 0-2: sustain
			0: none 
			1-6: longer intervals
			7: until overridden by another command
		Bit 3-5: release, encoding identical to attack
		Bit 6-7: ignored
Byte 7		Group and unknown flag
		Bit 0-4: Group, 0x00 = all bracelets, 0x01-0x1f: bracelets in specified group
		Bit 5: unknown flag, related to random color in multicolor mode
		Bit 6-7: ignored

danielweidman, 2022-11-11:
Hi Sean,

Thanks so much for your thoughtful message and the info.

I did not try CMY--I'll have to do that. I didn't know it was a common standard in the industry.

I also didn't consider how the BT LE bracelets might share a similar command structure to the IR ones. The bit/byte descriptors you provide do sound like they could be very relevant, though I think there'd have to be at least some differences because the count of the bits doesn't quite line up, and I believe I tried changing some single bits within the second, third, and forth bytes and produced no visible results rather than colors with slightly modified red, green, and blue components. I haven't looked super in depth at the encoding patterns recently though, sometime I should sit down and compare.

Right now I am travelling and don't have my bracelets or IR blaster, but I'll look at this more when I can.

Thanks again,
Dani

sean1983, 2022-11-12:
Thanks for your reply Dani,

I will post some update on here with some discovery's.

But judging by the DMX channels by Harman/Onyx and also the DMX software I
use Avolites future's profiles I seems the is multiple things going in.

First off there is 15  Main Colours

RED
ROSE
MAGENTA
PURPLE
BLUE
LIGHTBLUE
CYAN
AQUA
GREEN
LIME
YELLOW
AMBER
WHITE 12.5%
WHITE 25%
WHITE 50%
WHITE 100%

I've matched these with most of the ones in your code.

Then there is also 15 Background Colours aka What you called DIM_

DIM_RED
GREEN_DIM
DIM_BLUE
LAVENDER
ULTRAVIOLET
WHITISH !?
WHITISH !?

Some of these names are from the fixture definition from Avolites & Onyx
Which also include the RGB values so I can at least see on screen if they
look similar.

Few of these seem right, gotta get my IR and

The wristbands don't seem to have a Intensity Chsnnel for RGB,  Just the 15
Bright and 15 Dim. Nor do them have an overall Intensity(Dimmer).

They look like they just Fade In/out times and Blackout to handle dimming.
Which makes sense,  if you were to hammer them with DMX they batteries
would be dead in no time, So having preset Colours, Fades, Effects & the
ability to sent a hand full of small stacked commands makes sense.

But then this were it gets interesting,

There is more than just Colour Code and Tail Code.  You might have noticed
some thet seem to be longer.

You can send a command via DMX that can define 3 colours,  Colour 1, Colour
2 and Background Colour.

Using other commands,  The bands can be told to Strobe, Fade, Pulse etc
between the 2 colours and rest at  background  colour.

I think this all happens in the Main IR code with the colours.

The tail code(s), multiple will register Fade In, Fade Out and Effect
Sustain.

At the Fade out section the background colour can be defined also.

Then there is also Fx Fuctions  These could be the Weird Codes,  With
settings such as,

 Set Color RGB
 Set Layer Active
 Set Layer Group
 Set Random Scale RGB
 Set Envelope Delay Out
 Set Envelope Iteration
 Set Signal Timeouts

Lots parameters can be set via these,  might explain the Weirdness.

The problem with there parameters that can be send it,  They are not
viable.  So you try to brute,  You don't see a response so have no idea
what has been chsnge if anything.

So for every command.you throw at them, you'd have to run though every
known command again to see if there was any change.

So I'm wondering,  I might be able to get the scope onto one of the pins
and see if i can find somewhere they react to Good or Bad commands.  Even
if it's a slightly change in voltage.

Anyways,  I am not really good with the coding and decoding,  but I know
DMX pretty well,  an lots about Pixels and Theatrical Lighting so Analysis
and experimention is key for me

That is a bit of read for your travels lol

Sorry I waffle on.  🙄

Thanks for all you've done,  I'm so happy with having these running as they
are,  can't wait find out when else is lurking inside that legged little
chip.

Enjoy the rest of your travel.

Sean

P.s. excuse any typos and bad grammar,  I've got bad eyes.

danielweidman, 2022-12-14:
Interesting! Thanks for all the research you've put into this, it's awesome. I'm sorry I haven't had more time to mess around with this stuff recently. You seem to have a pretty good idea of coding and decoding from the sound of it, haha.

You have a really good point about us not knowing whether or not the codes from the brute force that don't produce a visible effect are "doing anything". I imagine you're correct, many are probably doing stuff behind the scenes like changing those parameters.

Did you ever try things out with the scope?

sean1983, 2022-12-15:
> Interesting! Thanks for all the research you've put into this, it's awesome. I'm sorry I haven't had more time to mess around with this stuff recently. You seem to have a pretty good idea of coding [...]

Hi Dani,

No I haven't tried scoping it yet.

But I might not be as correct on the colors, There is alot more colors than I expected.

I have been running your Brute Force Script for the past few weeks and I've extracted nearly all the Colour Codes out of the Wristbands.  Got a few more to do,   But so far I'm at around 900 codes.  I think the Full Color Pallet will be 1024 colors in total.  32 Hues  by 32 Shades.

My next is the issues of sorting the color codes into the correct orders of hue and shade.  

Here is how I have started to do it so far.  I got a Color Sensor attached to another Arduino that outputs to the color via serial in RGB values 0-255. Then have the sensor strapped to the Wristband.  It works great.  So using the Brute Script every time it fines a code and I pause it and hit the Copy button a butch of times to get a stable reading. Then I've been copying the result from my terminal to the text file I am storing the codes from the brute adding the RGB values after each code in ##Comments.

I've done this method for the last 60 codes I've gotten,  but I'm stuck with all the 900 ones I've got out of the bands previously,  I don't want to sit for another couple of weeks re-bruting.

Not being a coder is the problem.

I could do with a script like the Brute Script that can read the txt file that is in the format outputted by the brute tool, and for each code,  will hold the Color On for say 5 seconds, and say after 2.5 seconds of the 5,  Read the second COM Port and get the RGB Value from the sensor, then amment the text file with the RGB Values again in the comments. Then once written resume to next code in the text file and so on.

Do you think you could help with this?

I dont think it would be all that hard to put as essentially it work work very similar to the Python Sender,  with bit extra to freeze on each color,  and read from 2nd serial device and output the text ammendment.

The idea of sorting all these codes is,  once sorted in correct color order the the pattern in the binary codes might start to make sense two what the IR values actully are.

danielweidman, 2022-12-15:
Hi again,

Wow, that's really awesome that you've found 900 colors! So many more than the 85ish base colors I did (assuming they are all base colors--are they? Or do some fade in/out or do probabilistic things?).

To confirm, you've already got 900 codes in a text file, and you want a script that will go through each code, and:
1. Repeatedly send the code for 5 seconds (to hold it on)
2. 2.5 seconds into sending the code, check a second COM port for some data
3. Write both the code that was transmitted and the data received from the second COM port into a second text file.

Is that right? If so, I should be able to help with it.

How are you reading the data from the color sensor right now? And what's the model of the color sensor? If it works by sending raw data into a COM port as you suggest then that's awesome. I would have expected something like that to require some arbitrary software to read the values and maybe not even show up as a COM port on Windows, but I've never owned a color sensor before.

One thing though regarding the color pallete size: I saw on the PixMob site the following:
> With our palette of **16 million colours**, our pixels can beat to the rhythm of a pop song, gently shimmer to classical music, pulsate like a heartbeat or light up your whole audience as one unifying the whole venue.
16 million colors would suggest 24-bit/3 byte colors like you mentioned in your first post.
Does your 1024 estimate number include things like different brightnesses or other variation? If there are many different brightnesses and variations on top of that then I guess that could reach 16 million (but there'd need to be a lot). Or, perhaps there are multiple types of packets with different lengths or something, and the ones we are investigating so far are "shorter" ones that don't have as many colors. Or perhaps they are being cheeky and counting multiple different effects like fade, etc as different "colors".

danielweidman, 2022-12-15:
Never mind my question about how your color sensor works--I must have missed or misinterpreted what you wrote the first time (that it's a module connected to a second Arduino which outputs the values be serial). That's perfect.

sean1983, 2022-12-15:
I will come back to you in a while as I'm out at the moment,   but I will leave you with these pictures in the mean time.

sean1983, 2022-12-16:
> assuming they are all base colors--are they? Or do some fade in/out or do probabilistic things?

I think I only came across 1 code that held for a longer duration,  all the codes founds where all the same length as far as I noticed anyway,  None seemed to be longer than a short Pulse.

With in all these codes,  when I say 900,  that includes all the different brightnesses

> To confirm, you've already got 900 codes in a text file, and you want a script that will go through each code, and:
> 1. Repeatedly send the code for 5 seconds (to hold it on)
> 2. 2.5 seconds into [...]

Exactly Right,  Spot On...

I probably could be done on one microcontroller,   I use ESP32 for my IR Blaser with your ESP code,  (Well I use ESP32s for everything I love them),  But with the Colour Sensor i am using it refused to work with my ESP32s,  I spent hours messing around with it just giving me garbage text output over serial,  tried everything,  different pins,  baud rates etc but it wad a no go,  but I had a Arduino UNO laying about and it worked first time on that.

So 2 MCUs works fine,  So IR TX via COM3,  Colour RGB Value RX on COM4.

I know you vito'd then next bit below,  but here ia a bit more info on colour sensor.

The Model is GY-33 of the PCB,  But it is an standard TAOS TCS34725 sensor that all readily available for hobby or industrial use,  but this GY-33 PCB not only uses that package but comes with LED illumination, (I disabled that), But multiple different data pins for different connectivity UART TTL or I²C, but also I nice cowl/len mount part on the front which sits nicely around the LED sections on the wristbands.

It actully has it own MCU onboard for use via UART that will give average colour information but I've just got it set via jumper to I²C as it's the raw data output and it is apparently more accurate.

The sensor has a IR Filter which is great for this use since I am firing IR all over the place to trigger the bands,  plus eliminate wrong colours from the tiny amount or IR emmited from the LED Packages them selves, 

It detects 4 different light values,  RED, GREEN, BLUE, CLEAR.

Clear being White/Mixed spectrum, I believe it uses the Clear Light sensor along side the RGB Sensors with some math to calculate Colour Temperature and LUX,

But as we are working with RGB LEDs, this Clear value isn't needed.

I am using an example sketch I found for this model online that allows me to disable the  illumination LEDs, then I have modified the code to take the 16bits value 0-65535 per colour channel to 8bits 0-255  per channel, 16bit would give just over 196 colour variants,  and that gives me a head ache thinking about.  So 8bit values gives a maximum of 65535 colours,  which is a nice value to work with as it works nicely with things like DMX 0-255 per channel, or SImple Hex codes #000000 to #FFFFFF.

Then I have removed  everything else from the serial output,  So it just outputs a clean RGB value, 0-255, Comma Seperated. That just outputs on a new line in a loop.

0, 0,0 > 255, 255, 255

> ~~How are you reading the data from the color sensor right now? And what's the model of the color sensor? If it works by sending raw data into a COM port as you suggest then that's awesome. I would [...]

> One thing though regarding the color pallete size: I saw on the PixMob site the following:

If you look at the DMX channels layout they have for the Conteollers, there is only 3 color channels,  R, G, & B,  With DMX each channel can be adjusted between 0-255 so across the 3 RGB channels, 768 values, minus the Zeros as they are off (No Light) leaving 765 possible colours.

Now take my just shy of 900 codes,  I'm pretty sure there is going to be a few duplicates in there,  I bet it will come somewhere around that number of 765

> 16 million colors would suggest 24-bit/3 byte colors like you mentioned in your first post.

I am doubting 16.7million colours,  24bit  colour is a bit extreme,  RGB LEDs sre not exactly very good at representing 16bit colour specially at the price point these are probably at,  Looking at the LED Chips they are very generic looking SMD Packages   compared so some of RGB scripts I have rolls of,  they looks exactly the same,  Cheap Chinese RGB 5050 Packages,  

Plus they are using DMX to control these,  DMX is 256 values per Channel, 512 Channels per Universe,

So 16.7Millions colours would end up as 128 DMX Universes thats of 65535 DMX Channels each with a value to 0-255.

For addressing colours to wrist bands this would be silly and not practical and a waste of bandwidth if attempted over Artnet or E1-31 (DMX-Over-Ethernet),  There is other protocols used for this kinda of colour resolution generally used for Large Matscies,  But your moving from the realm of Lighting into the world a Video then.

There is more factors that make me sway away from this idea,  think about the tidy 8 legged mcu in the writeband, Holding all the data for such high colour resolution,  plus the processing of it, and the shear amount of IR Codes needed I don't think would fit in that burts of data we are seeing them use and all operating on a couple of AA's.

So I call B.S. for know,  but who knows  once we have the colours in order from the sensor data  we could see a pattern that may unlock more,

There could be a simple bit shift that make it jump to different pallet Hues, maybe a bit like how most of the codes via IR start 101,  but there is also the 110 ones that seem to shift the rest of the bit [... cut ...] 't look away for second incase I missed one,  but after that 700ish mark,  I changed the batteries and pulled the (New DIY) Pins out on 8 other wristbands I replaced batteries in a few weeks ago!  And surrounded myself with them so no matter where I looked I couldn't not see a flash,   because sometimes I was sitting for 20>40mins with the brute flying by without a single flash. 

I'll leave a screenshot teasers of the text file!

I will add everything to my film once I've tried up things a bit.

danielweidman, 2022-12-16:
Hi Sean!

It's very interesting to read your analysis, and I'm so glad you have such a good understanding of DMX control because I don't. I also really love ESP32s.

One more thing to confirm about how the script RGB-reader needs to work: You say `That just outputs on a new line in a loop.`
How frequently does it output? just super quickly or is there some delay or trigger?
(by the way, it's probably easier for me to implement with it done on two microcontrollers anyway so I can just separately look at the two COM ports instead of filtering between data for one).

Regarding color counts, you, mention having three channels (R, G, B), each with 256 possible values (0-255, an 8 bit [2^8] number). By my math, that actually does come out to >16.7 million colors:

`256*256*256 = 16777216` (minus 1 if you don't count 'off' as a color)

I think you have to do that instead of just:

`256*3=768` (minus 1 if you don't count 'off' as a color)

Since it's every possible combination of the three values. As a math example, if there were just 4 possible values (1-4), I'd say that's `4*4*4 = 64` (or 63) colors. I've attached a screenshot of a spreadsheet with an example--there are 64 rows with RGB combinations and there are no repetitions. Does that make sense? Or did I misinterpret what you were saying to begin with?

Also:
> There could be a simple bit shift that make it jump to different pallet Hues, maybe a bit like how most of the codes via IR start 101, but there is also the 110 ones that seem to shift the rest of the bits to different places.
That's an interesting though. I think I considered that briefly but never really looked too far into it. Part of me is scared that converting to binary in the way I did is potentially a huge red herring--I'm not sure how else it could be done though.

By the way, I have recently gotten some small, simple single-hard-PCB bracelets from an event (they didn't seem to be officially collecting them for recycling them and a few people had placed them in a random basket used by the security people for bag inspections; I asked the security people and they said I could have them). Would you like a few of those to test with? I can imagine two reasons that would be useful: 1) some codes I've found seem to either only work on the Coldplay bracelets, (or perhaps they only work on the group for the one Coldplay bracelet I own), and 2) the PCB is a lot simpler than the flexible ones on the Coldplay bracelets and might be easier to use with an oscilloscope.
[image]

sean1983, 2022-12-16:
Hi Dani,

First off,

> Regarding color counts, you, mention having three channels (R, G, B), each with 256 possible values (0-255, an 8 bit [2^8] number). By my math, that actually does come out to >16.7 million colors:

Well damn it,   It was like 3am here, and i messed up the math's badly!  Math's isn't my strong point anyways.  Your correct with the 256x256x256. Which means we're still short over 16.7millions colours,  It would be cool if we could find them,  But i think 32 Colours with 32 Brightness would be perfectly acceptable.

I'd be happy if i could get just enough for fulfill one of these colourwheels,

[image]

[image]

> One more thing to confirm about how the script RGB-reader needs to work: You say That just outputs on a new line in a loop.
How frequently does it output? just super quickly or is there some delay or trigger?

Maybe this would be easier,    Here's the code i'm using,

`#include <Wire.h>
#include "Adafruit_TCS34725.h"
#define commonAnode true

// our RGB -> eye-recognized gamma color
byte gammatable[256];

Adafruit_TCS34725 tcs = Adafruit_TCS34725(TCS34725_INTEGRATIONTIME_50MS, TCS34725_GAIN_4X);

void setup() {
  Serial.begin(9600);
  Serial.println("Pixel Colour Detect");

  if (tcs.begin()) {
    Serial.println("Found sensor");
  } else {
    Serial.println("No TCS34725 found ... check your connections");
    while (1); // halt!
  }

  for (int i=0; i<256; i++) {
    float x = i;
    x /= 255;
    x = pow(x, 2.5);
    x *= 255;
      
    if (commonAnode) {
      gammatable[i] = 255 - x;
    } else {
      gammatable[i] = x;      
    }
    //Serial.println(gammatable[i]);
  }
}

void loop() {
  uint16_t clear, red, green, blue;

  tcs.setInterrupt(false);      // turn on LED
  delay(100);  // takes 50ms to read 
  tcs.getRawData(&red, &green, &blue, &clear);
  tcs.setInterrupt(true);  // turn off LED
  
  // Figure out some basic hex code for visualization
  uint32_t sum = clear;
  float r, g, b;
  r = red; r /= sum;
  g = green; g /= sum;
  b = blue; b /= sum;
  r *= 256; g *= 256; b *= 256;

[... 41 lines cut ...]
 Instead i think this is done by how I mention in a earlier comment,   Basicly using 'probably a handheld' programmer,   Then say there is 10 Groups of wristbands for different parts of an Arena,  They could have there bulk containers holding all the bands for Group ID01, Then simply 1 shot IR Program the container for specific group, out of IR sight of the other containers.

But this is why this isnt going to work.....The battery tab's would still be in...

Now how i believe they actually do it is;

Once the arena is full and people have had the Tabs pulled out,  Then the they production team simply use their IR Moving Head Fixture, Focus in on Each individual Area they want to make a group, then IR Zap the Area with the Programming for Group, including Colour Pallet to use for the Special Effect,  i.e. the Dual Colour Fade Effects,  Dual Color Strobe Effects, & Multi-Colour Randoms.  Note:  The only time so far were we have noticed a difference between the colours is the Dual Colour FX being different when its comes to Coldplay vs The Weeknd.    

There Programming would be done by enabling the Programming DMX Channel 22 'Programming' Value 128>255 'Yes' then using the values within Channel 18. 'FX Ctrl'

[image]
[image]
[image]

**We could run into issues if we came accross these codes by mistake**

Using the moving heads would make a lot of sense as with modern moving head fixtures you can move to very accurate positions,  zoom to an exact area,  bring flats into the beam to shape it to the precise shape of the target area.  Then once the groups are programmed they can then just use the IR Wash fixtures to cover the whole arena and ones the groups who ID's are being sent will respond.

They use the moving heads in another interesting way at Coldplay,  and that is by using a GOBO (A Shaped Disk Silhouette) to shape the IR so that only Wristbands within its focus fire up and cause the shape be shown across the crowd,  In the case of Coldplay this was Red Shimmering Hearts.

But there is also the Video option,   I'm not sure exactly what this is,  It might be an alternative to the GOBO approach or might be what proves my last statement incorrect, Rather than using GOBO's,  The moving head needs a lot of power,   I have recently seen some modern LED moving heads inside,  The Light Source ('Light Engine' as they seems to call these EXTREMELY powerful LED arrays) they using is hundreds of tiny SMD LED's tightly jammed right up to each other on a heat sink,  The Martin Brand fixtures i saw inside of could using the array of LED's as a Matrix Display and project video,  Very low resolution mind,   I think it is something like 320x320,  But i don't see what PixMob couldn't do the exact same thing with  only IR rather than RGBUWW.  I would guess what they are using is like fixtures from a big brands that are modified to be IR.  I wish i could fine a Manual or Datasheets for the Movers like the one i have the the Wash Fixture.

Anyways,  I'm going off on one again!

I'll leave that with you,  Hopefully that code helps!

danielweidman, 2022-12-22:
Hello Sean!

Sorry for the slow response (again).

I wrote two scripts to try and use with your color sensor. One does the thing where we repeatedly send one color command for a few seconds at a time and read from the color sensor simultaneously, and one just reads right after sending each color. The former will probably work better, but I also made the latter in case there's a bug or something and it happens to work okay without the repeated sending.
See: [ir_code_discovery_and_export/send_and_read_color_detector_repeater.py](https://github.com/danielweidman/pixmob-ir-reverse-engineering/blob/main/ir_code_discovery_and_export/send_and_read_color_detector_repeater.py)
And: [ir_code_discovery_and_export/send_and_read_color_detector.py](https://github.com/danielweidman/pixmob-ir-reverse-engineering/blob/main/ir_code_discovery_and_export/send_and_read_color_detector.py)

> Which means we're still short over 16.7millions colours, It would be cool if we could find them, But i think 32 Colours with 32 Brightness would be perfectly acceptable.

32 colors and 32 shades would certainly be excellent. I'm surprised the Pixmob people even bothered to support so many more than that--but I guess it probably didn't add much complexity.

> I would definitely be interested in them, If that would be okay mailing to the UK, But before you do, Are they anything like these?
> Does it have a version Number or Name ? What kind of Cells does it take ? and how many LED Packages does it have on the PCB ?

The ones I brought up say "VIC v2.3r1" and are dated "20211206". Interestingly though, I also have a couple PALM v2.1 bracelets, with PCBs that look pretty similar to your Spice Girls PALM v1.7 one but not identical (and the date is a lot later). I'm attaching photos of both. Both have take 2 CR1632s and have 2 LED packages that are not quite as good at color rendition as the Coldplay ones, haha.
[image]
[image]

I would be happy to send some to the UK

I've been pretty surprised to find that most of the colors and tail codes (though not all) work on the smaller and older ones (pre-2018) as well. There are some interesting codes that only work on the older ones (motion sensitive ones which of course need the hardware the newer ones are missing but also some codes that seem to be able to be used for flashing between different colors at variable rates through repetition of the same code).

> Initial Thought was: (Probably wrong, I'll explain below)
> Instead i think this is done by how I mention in a earlier comment, Basicly using 'probably a handheld' programmer, Then say there is 10 [...]

I feel like if I were designing it and just needed positional grouping, that's exactly what I'd do (the moving head thing). (as an aside, I'm disappointed that it seemed like they didn't use any positional-dependent effects at all for the Weeknd concert or Cavs game I went to recently).

One thing relevant (and perhaps slightly contradictory?) to what you mentioned about sending commands for groups of bracelets in different containers may be this [job listing I found a while ago](https://www.recruit.net/job/production-worker-jobs/259F222B3C82ABB3). It lists as one of the job responsibilities:
`* Flashing, “grouping” and pre-”grouping” of our wearables`
Also, I've been surprised to find that at the last couple events I got bracelets at (the only ones I have a clear memory of), the pull tab was pulled by the time I got the bracelet. It looked like all the ones in the the boxes the people were handing them out from already had the tabs pulled/maybe never had tabs in the first place? Not sure how that would work exactly. I guess there could be some super deep sleep mode that would enable them to last for a few months/years only sampling IR very infrequently? Seems too good to be true though. I can't imagine how tedious it would be for the staff to go through every bracelet and thousands of tabs before each event but I guess it's possible (and maybe part of the above-linked job). They could definitely last at least a few days with the tab pulled and still work for a whole show.

At the events you've been to, were the bracelets distributed at the venue door, the section entry door, or left at the seats? I've only ever gotten them at the venue entry door--of course that would preclude any preprogrammed positional grouping since you can't know where guests will sit.

> But there is also the Video option, I'm not sure exactly what this is, It might be an alternative to the GOBO approach or might be what proves my last statement incorrect, Rather than using GOBO's, [...]

Interesting! My theory on that was that they could have a bright IR light source that blinks out the relevant commands, and then they put a high-contrast LCD in front of it (so the IR light source takes that place of a "backlight"). I think that's how an LCD projector works--just with a constant-on white visible light "bulb" for the light source instead of a blinking set of IR leds. Now that I think about it, there's no reason it could work like a DLP projector either--with the same light source difference and no color wheel. Does that make sense?

sean1983, 2022-12-22:
Hi Mate, 

Dont worry about fast replies,

Cheers for the scripts,  I will give thema try tomorow moring and let you know how i get on,

> I'm surprised the Pixmob people even bothered to support so many more than that--but I guess it probably didn't add much complexity.

I am sure it will be for the reason of Standards,   They can be used for many use's not just concerts, But during concerts the is normally Camera,  Video  for live use or recordings at then i comes to video coloir accuracy is very important,  But with it being 255x255x255 channels i would guess its normal practice as with any other light mixing via DMX.

> I would be happy to send some to the UK

Definety interested,  

I would love to fine out what General 8 Pin Microcontroller is,  I bet not a lot has changes, The PCB's are so simlar not alot changed, Extrta couple resistors but thats about it.

> I've been pretty surprised to find that most of the colors and tail codes (though not all) work on the smaller and older ones (pre-2018) as well. There are some interesting codes that only work on the older ones

Yes these 2 Spice Girls ones dont react to some of the tail codes too. Im suprised i didnt come access more tail codes during brue but then again,  They might no do anything as mentioned previously.

I am sure the motion activated ones will be the same software,  same chip, well pretty much same design but with a pads for the viberation switch can to solder down to,   I was thinking it could access 2of the 3 pins replacing the IR which a change in firmware,  But remembered they exist with IR and Motion.

Woud love to know what pins on the MCU it would use,  Having a switch could unlock hidden possibilities.

> I'm disappointed that it seemed like they didn't use any positional-dependent effects at all for the Weeknd concert or Cavs game I went to recently).

I think Coldplay are probably there biggest or maybe favorite client,  Coldplay like to go All Out when it comes to Lighting, Pyro and Visuals and PixMob probably wouldn't exist it it wasnt for the coldplays Xylo-Band's.

> One thing relevant (and perhaps slightly contradictory?) to what you mentioned about sending commands for groups of bracelets in different containers may be this job listing

Well,  Yes programming them would have to be via IR as there is no other input,  No Pins or Pads,  But i let you the grouping is probably be 1% of that job, And the other 99% will be popping the bands open with a jjg then flipping Coil Cells or AAs,
$17pa for replacing thousands of batteries,  errrm  I'll pass!

I dont think they will do much Grouping like this as much these days as the moving head makes this much eaiser.

> At the events you've been to, were the bracelets distributed at the venue door, the section entry door, or left at the seats?

Both Coldplay Gigs i went to this year the were giving them out at the enerance after security,  but also outside on the street/approach to arena,  The ones from big semi-white plastic bins (Looked like a small IBC Container) where give out inside at the entrance and the Tabs where pulled,  The ones give out at the street where pulled from big while plastic sacks at the tabs where felt in,  At whembly they people handing them out from the plastic container was pulling the tabs as passing them too you!
 
But in all cases adove,  I took well more than i was ment to,  but sneakiler weaving!  Plug being visually impaired and being guided in a special enterance give me an extra opertunity to get more,   After the Glasgow Coldplay get I left the venue with 9 of them.

Pulsing my IR Remote blaster i had at the big containers when leaving was VERY AMUSING for me!  But pretty confusing for the door staff collection the from people.

Another not from the last coldplay gig,  That peskie IR gets everywhere,  I went the toilet and there was at least 3 walls between me and the arena event space and the bloody thing was still working happily,  it wasnt until i covering it with my huddie it stopped.

But yes,  I got a few Tabbed ones,  and no different in operation.

> I've only ever gotten them at the venue entry door
Ive also being buying them from eBay and Facebook Marketplace when price is a fiver or less, and a few of these where tabbed still.

> Interesting! My theory on that was that they could have a bright IR light source that blinks out the relevant commands, 

LCD wouldnt work,   Its cant contrast enough to block that kind of light intensity,  TBH its cant hardly work well enough on monitors.   This has been tried in the Lighting Industy,  A couple of companies tried to make moving heads thats where just essentially an LCD Projectior but with just I mush larger HMI Lamp.  They where having to use 1200watt HMI lamps the get the equivalent of a 250watt HMI Standard Gobo Fixture.

DLP could possibly be a OK solution,   but it does have terrible goasting and dift issues,   I am watching DLP right now and i luckily dont see that colour shifting or rainbow still people get from it,  but my phone as see it fine,   But that drifting might well mess with the IR signal if used for GFX or Video with an IR Code.

Where modern LED Light engines are already a dence Matrix of LED that are individually controllable for Framing, Digital Gobos, Video,  but also for Longevity and evergy saving.

Excuse any typos, gramma or non-sence,  ive been away over 24 hours!

sean1983, 2022-12-23:
Hi Mate, 

> I wrote two scripts to try and use with your color sensor. One does the thing where we repeatedly send one color command for a few seconds at a time and read from the color sensor simultaneously, [...]

I've got "Im not a coder error"  (Probably something obvious)

`Traceback (most recent call last):

  	File ".\send_and_read_color_detector.py", line 67, in <module>
    color_detecter_res = read_recent_color_detector_line()
    
  	File ".\send_and_read_color_detector.py", line 37, in read_recent_color_detector_line
    if color_read_line.count(",")  < 2:
    
	TypeError: argument should be integer or bytes-like object, not 'str'

It flashing LED's first colour in my list,  then drops out with this error,  same with the other script.

Its problably something simlar!

Cheers

Sean

danielweidman, 2022-12-23:
Hey there! I want to respond about your other comments (and send you an email), but real quick about the script:

Try commenting out/removing lines 38-40 in the non-repeater version and lines 39-41 in the repeater version and let me know what happens! Those lines are probably not necessary.

sean1983, 2022-12-23:
> Hey there! I want to respond about your other comments (and send you an email) later, but real quick about the script:
> 
> Try commenting out/removing lines 38-40 in the non-repeater version and lines 39-41 in the repeater version and let me know what happens! Those lines are probably not necessary.
> 
> Also, I assume you set the COM ports for the two Arduinos in the code, but just a reminder if you didn't.

Okay, This works but the output is wrong, But i think i know why!

Here is the output from the 2 text files,   1st is from normal script,  2nd from repeater.

`
"RED":	[1, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 0, 0, 0, 1, 0, 0, 0, 1, 1, 0, 1, 0, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 1, 0, 0, 0, 0, 1], //b',0\r\n'

"GREEN":	[1, 1, 0, 0, 1, 0, 1, 0, 1, 1, 0, 0, 0, 0, 1, 0, 0, 1, 0, 0, 1, 0, 0, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 1, 0, 0, 0, 0, 1], //b'\n'

"BLUE":	[1, 0, 1, 0, 0, 0, 1, 1, 0, 1, 0, 0, 0, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1, 1, 0, 1, 0, 0, 1, 0, 1, 0, 1, 0, 1, 0, 0, 0, 1], //b''

"CYAN":	[1, 0, 1, 0, 1, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 0, 0, 1, 0, 1, 1, 0, 0, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 0, 1, 0, 0, 0, 1], //b''

"MAGENTA":	[1, 0, 1, 0, 1, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 0, 0, 0, 1, 1, 0, 0, 0, 1, 0, 1, 0, 1, 1, 0, 0, 0, 1, 0, 1, 0, 0, 0, 1], //b''

"YELLOW":	[1, 1, 0, 0, 1, 0, 1, 0, 1, 1, 0, 0, 0, 0, 1, 0, 0, 0, 1, 0, 1, 0, 0, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 1, 0, 0, 0, 0, 1] //b''
`

`
"RED":	[1, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 0, 0, 0, 1, 0, 0, 0, 1, 1, 0, 1, 0, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 1, 0, 0, 0, 0, 1], //b'Pixel Colour Detect\r\n'

"GREEN":	[1, 1, 0, 0, 1, 0, 1, 0, 1, 1, 0, 0, 0, 0, 1, 0, 0, 1, 0, 0, 1, 0, 0, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 1, 0, 0, 0, 0, 1], //b'Pixel Colour Detect\r\n'

"BLUE":	[1, 0, 1, 0, 0, 0, 1, 1, 0, 1, 0, 0, 0, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1, 1, 0, 1, 0, 0, 1, 0, 1, 0, 1, 0, 1, 0, 0, 0, 1], //b'Pixel Colour Detect\r\n'

"CYAN":	[1, 0, 1, 0, 1, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 0, 0, 1, 0, 1, 1, 0, 0, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 0, 1, 0, 0, 0, 1], //b'Pixel Colour Detect\r\n'

"MAGENTA":	[1, 0, 1, 0, 1, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 0, 0, 0, 1, 1, 0, 0, 0, 1, 0, 1, 0, 1, 1, 0, 0, 0, 1, 0, 1, 0, 0, 0, 1], //b'Pixel Colour Detect\r\n'

"YELLOW":	[1, 1, 0, 0, 1, 0, 1, 0, 1, 1, 0, 0, 0, 0, 1, 0, 0, 0, 1, 0, 1, 0, 0, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 1, 0, 0, 0, 0, 1] //b'Pixel Colour Detect\r\n'
`

Here is what is happening,

"Pixel Colour Detect" is coming from my aurdino with the sensor, I noticed it in the code from that sketch.

For some reason why i Monitor the serial output it must cause must cause it to restart, so it prints

`Pixel Colour Detect

 Found sensor 0`

I will remove these from that sketch so it doesnt print them,  I dont know why it restarts when new session connects though.

danielweidman, 2022-12-23:
> Okay, This works but the output is wrong, But i think i know why!
>
> Here is the output from the 2 text files, 1st is from normal script, 2nd from repeater.
>
> ` "RED": [1, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, [...]

Interesting, okay. I hope that works.

If it doesn't, the script could be re-written to open the serial connection once and keep it open. That would probably be better.

It looks like it's reading as bytes instead of a string, so adding a `.decode()` or after `.readline()` (so it would be `ser.readline().decode()`) would be good--it should get rid of the `b`s but not have anything to do with the main problem.

This is sad because I feel like it would be fast to interate in-person/if we both had the same hardware, but alas...

sean1983, 2022-12-23:
Okay Partial Success,

send_and_read_color_detector still Outputs,

"RED":	[1, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 0, 0, 0, 1, 0, 0, 0, 1, 1, 0, 1, 0, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 1, 0, 0, 0, 0, 1], //b',0\r\n'

"GREEN":	[1, 1, 0, 0, 1, 0, 1, 0, 1, 1, 0, 0, 0, 0, 1, 0, 0, 1, 0, 0, 1, 0, 0, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 1, 0, 0, 0, 0, 1], //b'\n'

"BLUE":	[1, 0, 1, 0, 0, 0, 1, 1, 0, 1, 0, 0, 0, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1, 1, 0, 1, 0, 0, 1, 0, 1, 0, 1, 0, 1, 0, 0, 0, 1], //b''

"CYAN":	[1, 0, 1, 0, 1, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 0, 0, 1, 0, 1, 1, 0, 0, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 0, 1, 0, 0, 0, 1], //b''

"MAGENTA":	[1, 0, 1, 0, 1, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 0, 0, 0, 1, 1, 0, 0, 0, 1, 0, 1, 0, 1, 1, 0, 0, 0, 1, 0, 1, 0, 0, 0, 1], //b''

"YELLOW":	[1, 1, 0, 0, 1, 0, 1, 0, 1, 1, 0, 0, 0, 0, 1, 0, 0, 0, 1, 0, 1, 0, 0, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 1, 0, 0, 0, 0, 1] //b''

I've added the blank lines so it formats better on here,

But yes,  With the fast script,  Serial port would need to be opened, then some time given,  then start reading,  While all the codes are sent,  then finally closed.

But with the reapeater script,   It just needs some tidying up,  Maybe on the way i am outputting from my Arduino,

But is the output from Repeater,

"RED":	[1, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 0, 0, 0, 1, 0, 0, 0, 1, 1, 0, 1, 0, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 1, 0, 0, 0, 0, 1], //b'251,32,37\r\n'

"GREEN":	[1, 1, 0, 0, 1, 0, 1, 0, 1, 1, 0, 0, 0, 0, 1, 0, 0, 1, 0, 0, 1, 0, 0, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 1, 0, 0, 0, 0, 1], //b'117,249,132\r\n'

"BLUE":	[1, 0, 1, 0, 0, 0, 1, 1, 0, 1, 0, 0, 0, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1, 1, 0, 1, 0, 0, 1, 0, 1, 0, 1, 0, 1, 0, 0, 0, 1], //b'13,114,252\r\n'

"CYAN":	[1, 0, 1, 0, 1, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 0, 0, 1, 0, 1, 1, 0, 0, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 0, 1, 0, 0, 0, 1], //b'155,251,219\r\n'

"MAGENTA":	[1, 0, 1, 0, 1, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 0, 0, 0, 1, 1, 0, 0, 0, 1, 0, 1, 0, 1, 1, 0, 0, 0, 1, 0, 1, 0, 0, 0, 1], //b'253,109,153\r\n'

Maybe the way i am printing could be improved,

`Serial.print((int)r, DEC);

Serial.print(",");

Serial.print((int)g, DEC);

Serial.print(",");

Serial.print((int)b, DEC);

Serial.println();`

danielweidman, 2022-12-23:
Oh okay! Did you add the decode() part? I would expect that to do most of the the cleaning up.

If you send the whole script as you are running it now I could add it and send it back.

sean1983, 2022-12-23:
> Oh okay! Did you add the decode() part? I would expect that to do most of the the cleaning up.
> 
> If you send the whole script as you are running it now I could add it and send it back.

Sorry,  I think i missed the bit where you mention the decode while i was debugging.

Here is the complete code

`#include <Wire.h>
#include "Adafruit_TCS34725.h"
#define commonAnode true

// our RGB -> eye-recognized gamma color
byte gammatable[256];

Adafruit_TCS34725 tcs = Adafruit_TCS34725(TCS34725_INTEGRATIONTIME_50MS, TCS34725_GAIN_4X);

void setup() {
Serial.begin(9600);

for (int i=0; i<256; i++) {
float x = i;
x /= 255;
x = pow(x, 2.5);
x *= 255;

if (commonAnode) {
  gammatable[i] = 255 - x;
} else {
  gammatable[i] = x;      
}
}
}

void loop() {
uint16_t clear, red, green, blue;

tcs.setInterrupt(false); // turn on LED
delay(50); // takes 50ms to read
tcs.getRawData(&red, &green, &blue, &clear);
tcs.setInterrupt(true); // turn off LED

// Figure out some basic hex code for visualization
uint32_t sum = clear;
float r, g, b;
r = red; r /= sum;
g = green; g /= sum;
b = blue; b /= sum;
r *= 256; g *= 256; b *= 256;

Serial.print((int)r, DEC);
Serial.print(",");
Serial.print((int)g, DEC);
Serial.print(",");
Serial.print((int)b, DEC);
Serial.println();
}`

danielweidman, 2022-12-23:
Oh okay! Regarding the .decode(), I was actually suggesting adding that in my scripts

sean1983, 2022-12-23:
> .readline().decode()

I did that,  Now the Output has gone!

"RED":	[1, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 0, 0, 0, 1, 0, 0, 0, 1, 1, 0, 1, 0, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 1, 0, 0, 0, 0, 1], //
"GREEN":	[1, 1, 0, 0, 1, 0, 1, 0, 1, 1, 0, 0, 0, 0, 1, 0, 0, 1, 0, 0, 1, 0, 0, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 1, 0, 0, 0, 0, 1], //
"BLUE":	[1, 0, 1, 0, 0, 0, 1, 1, 0, 1, 0, 0, 0, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1, 1, 0, 1, 0, 0, 1, 0, 1, 0, 1, 0, 1, 0, 0, 0, 1], //
"CYAN":	[1, 0, 1, 0, 1, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 0, 0, 1, 0, 1, 1, 0, 0, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 0, 1, 0, 0, 0, 1], //
"MAGENTA":	[1, 0, 1, 0, 1, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 0, 0, 0, 1, 1, 0, 0, 0, 1, 0, 1, 0, 1, 1, 0, 0, 0, 1, 0, 1, 0, 0, 0, 1], //
"YELLOW":	[1, 1, 0, 0, 1, 0, 1, 0, 1, 1, 0, 0, 0, 0, 1, 0, 0, 0, 1, 0, 1, 0, 0, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 1, 0, 0, 0, 0, 1] //

danielweidman, 2022-12-23:
Pfft, darn! Haha, I guess it's better without that then.

If you just run the script and get everything with RGB values added, we should be able to just find/replace to remove the extra characters, right?

sean1983, 2022-12-24:
Yes,  Find and replace would work,  but I think the stuff at the end is
related to my poor way of formatting the RGB values.

/n being a quite common command for new line,  I'm sure it's to do with the
way I've broken to RGB value pint up,  pretty sure if that way consolidated
some way it would be okay,  not 100% but could be the issue.

danielweidman, 2023-01-02:
Hello again!

The reason I was saying it's an issue in my code is because `b'` in python indicates the beginning of a string representation of a `bytes`-type object. I think it's actually a combination of that and just the fact that the `serial` package doesn't convert the received newline characters `\r\n` to actual newlines. I updated the code now--want to try again?

NTLS09, 2023-01-06:
Hi,

I'm really interested in what @sean1983 is doing. Finding over 900 codes is amazing. Don't think I can make a mobile remote with that many... Maybe if I group the colours together with all the effects, would be very time-consuming unless someone makes a script. I can't code either, which is a shame.

Interested in the wristbands @sean1983 got. I got some from the O2 from the Jingle Bell Ball concert and I think ours might be similar.

When both of you are able to get a working script to get codes, I can attempt a demo remote. `pixmob@ns09.ga` if you can.

> I think Coldplay are probably one of their biggest or maybe favourite clients, Coldplay likes to go All Out when it comes to Lighting, Pyro and Visuals and PixMob probably wouldn't exist if it wasn't for the Coldplay's Xylo-Band's.
> 
Capital like to do it as well, and gave out X4 wristbands which use AAA and are super bright. They did it at the O2.

> With the latest ones Coldplay use, they are really simple to open and replace the batteries. I use 2 blue plastic Pry Tools from an iFixit Kit in the little stots down each side, then press and all 4 clips open perfectly, then pop in New AAA's.
>
My one opened like that, I opened one clip at a time to make sure I didn't break them. Got a small scratch in the plastic but I think it will be fine. 
[image]

Cheers

sean1983, 2023-01-09:
> Hello again!
>
> The reason I was saying it's an issue in my code is because `b'` in python indicates the beginning of a string representation of a `bytes`-type object. I think it's actually a [...]

Hi Dani,

So..  No luck with the single shot script,  it just outputs blank results,

But.  The Repeater Script works Perfectly,

[image]

Yes the repeater script is slower,  But does seem to capture accurate result,  So I am happy with this,  It has frozen a couple of time,   But if i ^C and re-run it runs okay!

So i will start and run my new codes through it in batches incase of crash,  I am probably going to hook up wristband to one of my bench PSU's so that there is no deviation as batteries drain.

I have previously made an excel sheet that will show all the colours for each line using the RGB Value,  and also converts it in the CMYK & HSL/V values too for easy sorting.

I'll let you know how i get along.

sean1983, 2023-01-09:
> Hi,
>
> I'm really interested in what @sean1983 is doing. Finding over 900 codes is amazing. Don't think I can make a mobile remote with that many... Maybe if I group the colours together with all the [...]

Hi @NTLS09 

The idea at the moment is to use my colour sensor to convert the codes i've found to RGB values,   once this has been done the colours can be sorted by the values and we might be able to a pattern in the IR Code that relates to the colour output, brightness/shades for each colour.

If a pattern is discovered we wont need the 900 codes,  just a few codes that we can manipulate to adjust possible 65535 colours these wristbands may contain.

The Capital FM Event bands look like the same as the latest coldplay ones,  minus the extra leds that are in the straps.

[image]

danielweidman, 2023-01-27:
I'm really glad to hear that the repeater script has worked! Sorry I have been busy and not active here. Good results?

danielweidman, 2023-02-03:
@sean1983 Hope you are doing well
In case you didn't see, @tnikitin just came across a new 2022 RF-based bracelet! https://github.com/danielweidman/pixmob-ir-reverse-engineering/issues/14#issuecomment-1410576543
Thought you might be interested.

KittinutzLivibe, 2024-02-22:
do anyone have code in side pixmob MCU ?