danielweidman/pixmob-ir-reverse-engineering #10: Using IR blaster from an Android phone

ibrunops, 2022-11-25:
Hello! First of all, excellent work. All the information in this repository is amazing, I would love to have a Flipper Zero and have recorded some IR broadcasts at a concert.
I wanted to ask you Daniel if it would be possible to use the IR blaster that comes in some android smartphones (for example, I have a Xiaomi Redmi Note 7) to reproduce the IR commands.
I found this app `https://play.google.com/store/apps/details?id=net.binarymode.android.irplus` that allows you to upload codes to be used by the IR blaster, but I'm having trouble figuring out how to use it. On their website `https://irplus-remote.github.io/#learning` they have a converter for example, and they explain how to use the app. Should I use the .ir files that are in the "raw_wild_ir_captures" folder? Because if I try to convert that code I don't get anything useful. I know I'm missing something but I don't know what it is, if you can help me that would be great.
Greetings and thanks in advance!

ibrunops, 2022-12-14:
Hi Daniel! Thanks for answering!
I can help you with whatever you need, if you can create an irplus file that would be great! Also if you want, we can coordinate a day in case, for example, you want to test some configuration of the 2nd app in real time with my phone, so we can discover it faster. My spoken English is not very good but I try my best haha

danielweidman, 2022-12-15:
Okay great! I'll see where I can get with generating an .irplus file, but that could be good.
Hopefully I will have an update tomorrow.

danielweidman, 2022-12-15:
I made an irplus file! I hope it works. It's just got one command for now that should make the bracelet blink red for a second or so. If it works I can add some more.

By the way, what show did you get the bracelet at? Have you replace the batteries since the event/how long ago if so?

ibrunops, 2022-12-15:
Hi Daniel! The irplus file works perfect!

Yes, my bracelet is from the Coldplay concert in november here in Argentina, it works fine, I had already made it work with your tutorial, using an Arduino Nano and a raw IR. I just had to change the batteries.

I was looking at the file and I understand that you used IrScrutinizer to create it. Did you have to read with that program the IR signal emitted by your python code with an IR receiver so that the program would then convert it to PRONTO_HEX and then to an irplus file?
Or did you do it another way?
I wanted to know because I would like to be able to make my own irplus files, but if it's like I said before, I would have to buy an IR receiver and the whole process is quite tedious, I thought there was a way to convert directly from your code.

danielweidman, 2022-12-16:
I'm so glad to hear the file works. I hope the Coldplay concert was good--I'm so sad they didn't come anywhere within a few hundred miles of where I live, haha.

I actually didn't need to use an IR receiver with IrScrutinizer at all! It was simpler than I thought, and I think you'll be able to do it yourself. It's still a little tedious because you have to do it code-by-code, but it's not that bad. You basically need to take the raw timing for the cleaned IR signals (which you can get from the .IR files here: https://github.com/danielweidman/flipper-pixmob-ir-codes. 

Here is what I did:

1. Install IrScrutinizer (https://github.com/bengtmartensson/IrScrutinizer)
2. Launch IrScrutinizer
3. Go to "options" at the top and UNCHECK "Invoke signal cleaner". This is important and tripped me up for a while.
4. Go to the "export" tab and set "Export Format" to "irplus"
5. Open one of the .IR files as text, maybe start with "PixMob_main.ir"
6. Got to one of the lines that starts with "data: ", and copy the numbers part, like `1400 1400 700 700 700 1400 700 2800 700 2100 1400 700 700 700 700 1400 1400 2800 1400 2800 700`. (Each of those numbers represents the number of microseconds until the signal switches from high to low or vice-versa).
7. Paste those numbers into the "Scrutinize signal" page
8. Press "Export"
9. Repeat steps 6-8 for each signal type you want to use with irplus.

Note: that will get you one file per signal/color effect/code. I think you might be able to get one file with multiple codes (presumably this would show as one remote with many buttons) another way: _Instead_ of pasting the signal into the "Scrutinize signal" page (step 7), go to the "Scrutinize remote" page and then the "Raw remote" tab, then right click somewhere in the tab and click "Import signal from clipboard". Hopefully you can repeat that over and over for each signal and it will only take a few minutes. I think that will give you a remote with multiple signals (you might be able to use the "Comment" column to name them). When you have added the ones you want to the list, press "Export". I didn't test this process though so no promises.

Does that make sense? I hope that the instructions are clear enough. Let me know how it goes.

(to be clear, I'm sure it would be possible to do this automatically with code, but it would require figuring out how the .irplus file format works and then writing a program and I don't have tons of time to spend on that right now.)

ibrunops, 2022-12-16:
Hi Daniel! You are a genious!
It's not tedious at all, it would be worse to do what I said with the IR receiver. I'm going to try it today, but I already know it works because you managed to do it! If I spare the time, I'll try to make a single irplus file with as many codes (effects) as possible, because the file structure doesn't seem that complicated. And as you say, in the app it will look like a remote control, I'll try so that instead of words there are colors directly in the app. And I will record a video showing the results.

A couple of things I wanted to know, each line that starts with "data" would be a color/effect, right? But it may not be something with immediate effect on the bracelet, is it? I would have to try all of them and see which works and which doesn't, in addition to categorizing them by color because it is not indicated in those files. I am correct, right?

If I can get a good understanding of the irplus files, maybe I could make some python code that would automate this, but no promises haha, but I'll try.

Another thing I wanted to tell you about, in the "brute_force_gui.py" file, the SKIP_FIRST_N_CODES variable is not initialized and when I try to run the program it gives me an error. I solved this by doing a SKIP_FIRST_N_CODES = 0
but I don't know what the variable does so I don't know if what I did was right, anyway doing that the program runs fine and I was able to find two new immediate effects, running as the program was written. I still can't quite understand which bits to change to find new combinations, if you could explain it to me again that would be great, otherwise don't worry.

Any news I have I will write it here, thanks again for everything!

danielweidman, 2022-12-16:
Hi there!

Thanks for the well-thought-out reply and questions. I'm about to go out
right now so I can't respond to everything, but I will say that if you use
the IR files in the "flipper" repository (rather than this one), pretty
much all of the codes will show an immediate effect. A few lines above the
data line there is a name for each:

https://github.com/danielweidman/flipper-pixmob-ir-codes/blob/main/PixMob_main.ir

The raw IR files in this repo are the ones that don't all have immediate
effects.

Also, setting SKIP_FIRST_N_CODES to 0 was good, I'll have to look into why
that was broken in the first place. I can explain what it does and answer
your other questions later!

- Dani

NTLS09, 2022-12-28:
Hi,
This works on the Capital Jingle Bell Ball concert bracelets from a month ago! Currently trying to use IrScrutinizer to modify more code. I'll let you know how it works.

closed 2023-03-25 by danielweidman

aguilerabr, 2023-04-08:
Guys,

I've converted some codes to irplus format. I'll publish how to convert .ir codes and convert it to .irplus format without using IrScrutinizer.

Copy the following text into a new empty file (notepad) and name it as [something.irplus] and then import to irplus application.

<irplus>
    
    <device manufacturer="Pixmob-Colors" model="PixMob" format="PRONTO_HEX">
	
        <button label="Red">0000 006D 000B 0000 0035 0035 001B 001B 001B 0035 001B 006A 001B 006A 001B 001B 001B 001B 001B 0035 0035 006A 0035 006A 001B 076C</button>
		
		<button label="Green">0000 006D 000B 0000 0035 0035 001B 001B 001B 001B 0035 006A 001B 0035 001B 0035 001B 0035 001B 0035 0035 006A 0035 006A 001B 076C</button>
		
		<button label="White">0000 006D 000C 0000 001B 001B 001B 0050 0035 001B 001B 006A 001B 0035 001B 001B 001B 006A 0035 0035 001B 0050 001B 0035 001B 0035 001B 076C</button>
		
		<button label="Turq">0000 006D 000C 0000 001B 001B 001B 001B 0035 0035 0035 006A 001B 0035 001B 001B 0035 0035 001B 0035 0035 006A 001B 001B 001B 0050 001B 076C</button>
		
		<button label="Blue">0000 006D 000E 0000 001B 001B 001B 0050 0035 001B 001B 006A 001B 0035 001B 001B 001B 0035 0035 001B 001B 0035 001B 001B 001B 001B 001B 001B 001B 0050 001B 076C</button>
		
		<button label="Magenta">0000 006D 000C 0000 001B 001B 001B 001B 0035 0035 0035 006A 001B 0050 0035 0050 001B 001B 001B 001B 0035 0050 001B 001B 001B 0050 001B 076C</button>
		
        <button label="Yellow">0000 006D 000B 0000 0035 0035 001B 001B 001B 001B 0035 006A 001B 0050 001B 001B 001B 0035 001B 0035 0035 006A 0035 006A 001B 076C</button>
		
		<button label="Pink">0000 006D 000C 0000 001B 001B 001B 0050 0035 001B 001B 006A 001B 0035 001B 001B 001B 006A 0035 0035 001B 0050 001B 001B 001B 0050 001B 076C</button>
		
		<button label="Orange">0000 006D 000C 0000 001B 001B 001B 001B 0035 0035 0035 006A 001B 0050 001B 001B 001B 0035 001B 001B 001B 0035 001B 0050 0035 006A 001B 076C</button>
		
    </device>
</irplus>

Hj2k1, 2023-09-14:
Hi man, could you make the ready-made irplu file available, I'm trying to do it the way you explained but I'm not succeeding, another question I noticed that your nickname is written _"BR"_ are you also from here in Brazil??

NexusXP, 2023-11-04:
> Hi man, could you make the ready-made irplu file available, I'm trying to do it the way you explained but I'm not succeeding, another question I noticed that your nickname is written _"BR"_ are you also from here in Brazil??

Hi, I managed to insert codes @aguilerabr provided into the codes @danielweidman provided by replacing the <button label to </devices> section with the new one. Here's the file: https://drive.google.com/file/d/1aq_BR_O_WG6dac8mDoNImq3yEFAI3rJn/view?usp=sharing

NLACE-COM, 2024-04-21:
Thanks for sharing the codes. With that, I designed a button panel in IRPlus that works very well on my Xiaomi phone.

Here it is: https://drive.google.com/file/d/1MDwlA1X-oaqCnGtmLwvYhDgckzV6VE1P/view?usp=drivesdk

theimpossibleheir, 2024-05-29:
Hello! Good day. Could anyone create those in google drive but make it constantly blinking in all colors. from all the file in the google drive, it was only remote to each color like blue, red, green, etc. in the it plus.

theimpossibleheir, 2024-05-29:
> I made an irplus file! I hope it works. It's just got one command for now that should make the bracelet blink red for a second or so. If it works I can add some more.
> 
> By the way, what show did you get the bracelet at? Have you replace the batteries since the event/how long ago if so?
> 
> Link: https://drive.google.com/file/d/17edwUHpE3IXD_OYrKe2ZmNfqv5JGiCCp/view?usp=sharing

can you make one like this in the video?

https://vt.tiktok.com/ZSY6YW7B5/

https://vt.tiktok.com/ZSY62d7na/

theimpossibleheir, 2024-05-29:
> Thanks for sharing the codes. With that, I designed a button panel in IRPlus that works very well on my Xiaomi phone.
> 
> Here it is: https://drive.google.com/file/d/1MDwlA1X-oaqCnGtmLwvYhDgckzV6VE1P/view?usp=drivesdk

can you make a constantly blinking like in this video?

https://vt.tiktok.com/ZSY6YW7B5/

https://vt.tiktok.com/ZSY62d7na/

theimpossibleheir, 2024-05-29:
could someone make a google drive link for it plus remote with this synchronization? https://vt.tiktok.com/ZSY62DxCj/

https://github.com/danielweidman/pixmob-ir-reverse-engineering/assets/171200591/5384ff30-058f-4106-86b2-842ebcff9c99

nikitaclicks, 2024-08-03:
For anyone exploring this, below is a file for irplus with all the signals from [PixMob_main.ir](https://github.com/danielweidman/flipper-pixmob-ir-codes/blob/main/PixMob_main.ir). Remove the `.txt` extension.

[pixmob_all.irplus.txt](https://github.com/user-attachments/files/16479110/pixmob_all.irplus.txt)

NTLS09, 2024-08-03:
Thanks, this is already on the repo in the MISC folder.

theimpossibleheir, 2024-08-06:
Hello. I just came across with shmmr in this website. https://beta.shmmr.app/register

I made my own shmmr device following all the instructions. 
[image]
[image]
[image]
[image]
[image]

uploading this arduino "into" file to the board. it does work
[ShmmrTransmitterFirmware_v1_2 (1).zip](https://github.com/user-attachments/files/16516640/ShmmrTransmitterFirmware_v1_2.1.zip)

MY PROBLEM IS, IT ONLY FLASHES ONE COLOR PERSISTENTLY LIKE THE COLOR BLUE OR GREEN OR RED OR WHITE. 

Can someone make a revision to the shmmr arduino "into" file to have loop effect continuely with all the colors since i don't really know how to do it? 
[image]
[image]

https://github.com/user-attachments/assets/8d08a191-d264-4730-9d49-065e251b8d8b

https://github.com/user-attachments/assets/7b636505-fea5-4c1c-bb41-bed60ec1956b

https://github.com/user-attachments/assets/05885455-22e2-467f-af74-6d7cac15767b

[image]
[image]

That's all the reference. This shmmr file does work. but i need someone to revise it for all colors like red, fade red, green, fade green, light green, fade light green, yellow fade, blue, blue fade, light blue, light blue fade, magenta fade, pink, white, tuquiose, and so on. 
[ShmmrTransmitterFirmware_v1_2 (1).zip](https://github.com/user-attachments/files/16516703/ShmmrTransmitterFirmware_v1_2.1.zip)

https://beta.shmmr.app/register

thank you very much

theimpossibleheir, 2024-08-06:
> Thanks, this is already on the repo in the MISC folder.
> […](#)

Hello. I just came across with shmmr in this website. https://beta.shmmr.app/register

I made my own shmmr device following all the instructions. 
[image]
[image]
[image]
[image]
[image]

uploading this arduino "into" file to the board. it does work
[ShmmrTransmitterFirmware_v1_2 (1).zip](https://github.com/user-attachments/files/16516640/ShmmrTransmitterFirmware_v1_2.1.zip)

MY PROBLEM IS, IT ONLY FLASHES ONE COLOR PERSISTENTLY LIKE THE COLOR BLUE OR GREEN OR RED OR WHITE. 

Can someone make a revision to the shmmr arduino "into" file to have loop effect continuely with all the colors since i don't really know how to do it? 
[image]
[image]

https://github.com/user-attachments/assets/8d08a191-d264-4730-9d49-065e251b8d8b

https://github.com/user-attachments/assets/7b636505-fea5-4c1c-bb41-bed60ec1956b

https://github.com/user-attachments/assets/05885455-22e2-467f-af74-6d7cac15767b

[image]
[image]

That's all the reference. This shmmr file does work. but i need someone to revise it for all colors like red, fade red, green, fade green, light green, fade light green, yellow fade, blue, blue fade, light blue, light blue fade, magenta fade, pink, white, tuquiose, and so on. 
[ShmmrTransmitterFirmware_v1_2 (1).zip](https://github.com/user-attachments/files/16516703/ShmmrTransmitterFirmware_v1_2.1.zip)

https://beta.shmmr.app/register

thank you very much

theimpossibleheir, 2024-08-07:
> For anyone exploring this, below is a file for irplus with all the signals from [PixMob_main.ir](https://github.com/danielweidman/flipper-pixmob-ir-codes/blob/main/PixMob_main.ir). Remove the `.txt` [...]

Hello. I just came across with shmmr in this website. https://beta.shmmr.app/register

I made my own shmmr device following all the instructions. 
[image]
[image]
[image]
[image]
[image]

uploading this arduino "into" file to the board. it does work
[ShmmrTransmitterFirmware_v1_2 (1).zip](https://github.com/user-attachments/files/16516640/ShmmrTransmitterFirmware_v1_2.1.zip)

MY PROBLEM IS, IT ONLY FLASHES ONE COLOR PERSISTENTLY LIKE THE COLOR BLUE OR GREEN OR RED OR WHITE. 

Can someone make a revision to the shmmr arduino "into" file to have loop effect continuely with all the colors since i don't really know how to do it? 
[image]
[image]

https://github.com/user-attachments/assets/8d08a191-d264-4730-9d49-065e251b8d8b

https://github.com/user-attachments/assets/7b636505-fea5-4c1c-bb41-bed60ec1956b

https://github.com/user-attachments/assets/05885455-22e2-467f-af74-6d7cac15767b

[image]
[image]

That's all the reference. This shmmr file does work. but i need someone to revise it for all colors like red, fade red, green, fade green, light green, fade light green, yellow fade, blue, blue fade, light blue, light blue fade, magenta fade, pink, white, tuquiose, and so on. 
[ShmmrTransmitterFirmware_v1_2 (1).zip](https://github.com/user-attachments/files/16516703/ShmmrTransmitterFirmware_v1_2.1.zip)

https://beta.shmmr.app/register

thank you very much