danielweidman/pixmob-ir-reverse-engineering #11: Imagine Dragons Signal Recordings

hckrdan, 2022-12-02:
Hi Daniel, great work, I just found your repo.
I went to the Imagine Dragon concert over a end of May 2022 weekend and recorded some IR signals using Scoppy app on my phone as a signal analyzer and an IR receiver connected to a raspberry Pi Pico - I haven't had a chance to reverse engineer the signals but I can upload them here if they would be of any use, the output files have a weird format as a table with time in a float format on the left column and signal level on the second column. Please let me know if you would like the csv's.
Cheers!
[image]
[image]
[image]
[image]

danielweidman, 2022-12-14:
Wow, that's so exciting and super crafty of you! I would really love to look through those CSV files and see if we can get them converted to a format that's easy to reproduce. This could be especially interesting since I imagine (get it?) that the recordings you made are long-running recordings, right?
The Flippers we've used for recording so far don't let you record for more than a second or so at a time--they try to guess when an IR transmission as "started" and "stopped", which is all well and good for getting button presses on a TV remote or something, but can be really limiting when it comes to trying to look at the results of a transmission consisting of many packets in order over a longer period of time.

Thanks so much for offering to share, I'd really appreciate it if you did.

Would you be able to either 1) put them in a Google Drive/similar folder and send a link or 2) 1) open a PR that adds these signals to a folder within the [raw_wild_ir_captures](https://github.com/danielweidman/pixmob-ir-reverse-engineering/tree/main/raw_wild_ir_captures) folder of this repo?

hckrdan, 2022-12-14:
Hi Daniel, I can try to upload to the repo tonight, I am not sure if these
were very long recordings either, the software has some limitations on how
many samples it would store (files are 1.39 MB CSVs), I'll share and you
can take a look, my pleasure to share but can't guarantee usability...

danielweidman, 2022-12-14:
Thanks very much!

hckrdan, 2022-12-15:
Try this and let me know, I also made a PR but gave up: https://drive.google.com/drive/folders/1wFtVZAicY7Os9OqLyASDs0Q3IeVvJql4

danielweidman, 2022-12-15:
Great! I'll get those added. I'll put your GitHub username in the README, but is there a human name you want included too? Fine if not.

danielweidman, 2022-12-15:
Got them added in, this is great. Soon (tomorrow hopefully) I'll try and make a converter from these files into signals of the format we can use with the Python transmitting code.

hckrdan, 2022-12-15:
Thanks Daniel, I appreciate it, I hope the recordings are usable. I was thinking to record some known signal (i.e. a Samsung remote power on/off button) for reference, I'll try that Friday or over the weekend.
-Dan

danielweidman, 2022-12-15:
So, I made a way to convert your signals to a form that I could retransmit...

I found that the packet structure does look pretty similar to what we've seen before for recorded packets, but sadly none of the signals that were recorded had an immediate effect on any of the bracelets I tried them with. So the signals could well be ones used to set up other effects/change bracelet settings/etc. Definitely still useful to have recordings of!

hckrdan, 2022-12-15:
Thank you Dani, I appreciate trying those signals, maybe my sampling of the
signals was not fast enough or they don't have the headers that they need
(signal analyzer may have recorded the backend of those signals) but was
worth a try!
Happy Holidays!