danielweidman/pixmob-ir-reverse-engineering #13: Pixmob irplus remote example

NTLS09, 2022-12-28:
I've made a pixmob remote using your guide on a previous issue. Testing on Redmi Note 9 with irplus and ir blaster. It's only got all colours and the first fade option for each. I prefer to use the fade as I have to keep pressing the colour, but it's a start. 

[https://drive.google.com/file/d/1sUoA17MCIWLifqiktEU4xMyKMIFIKq2j/view?usp=sharing](https://drive.google.com/file/d/1sUoA17MCIWLifqiktEU4xMyKMIFIKq2j/view?usp=sharing)

danielweidman, 2023-01-02:
Awesome! This is super cool. I wish I my phone IR blaster was compatible so I could try this out.

Would you mind if I added this file to the main repository (or you could do it via a PR)?

NTLS09, 2023-01-06:
You can add it. I'm no good in python, but I think if we can get a script that filters out their code and name, it would make it easier. Of course, if someone can program a python script to do so it would be great. Maybe we can reprogram a generic remote to send the correct codes, but that would be quite hard.

NTLS09, 2023-01-06:
I've been able to replace the batteries in it successfully. Mine uses AAA batteries. I'll attach photos. Ignore the textbook in the background, I'm using it to prop them up. It took some prying and a couple of scratches to open it, but I'm sure you can do a better job than I did as no one has opened these bracelets. They are the X4 bracelets from Capital Jingle Bell Ball in the O2 if you're interested. 

[https://drive.google.com/drive/folders/1F6Pur7nfx_bvHyjU7X2KIJQ2_B3mX0AU?usp=sharing](https://drive.google.com/drive/folders/1F6Pur7nfx_bvHyjU7X2KIJQ2_B3mX0AU?usp=sharing)

danielweidman, 2023-01-27:
Neat--thanks for the pictures! They are the first I've seen of that model with the batteries removed. I hope the concert was fun :)

I will add the remote to the repo and of course credit you (what name would you optionally like credited? in addition to Github username)--thanks so much for making it. It looks like the buttons are represented as PRONTO_HEX format and the file seems straightforward enough, minus layout stuff. How useful would a converter script be?

NTLS09, 2023-01-29:
Will add more info later, for the credit you can just put Nathaniel. i don't mind. will let you know if a script would be useful, not in a rush at all. I'll need to get my GitHub student profile done first and I'll let you know.
cheers

NTLS09, 2023-02-12:
I think I've found a way to import CSV or spreadsheet-like files to irscrutinizer, making a spreadsheet from the flipper file. I'll update it with the entire remote—no clue if it will work.

NTLS09, 2023-02-13:
Success! Made a remote with all colours and all effects on the flipper file. Bit messy, but it's fine for now. Cut down the flipper file, so that there only was the name and code. Then put the two together and separated with a comma. Imported to IR Scrutinizer and got a whole remote done. [Link](https://drive.google.com/file/d/1cBngqGHxFIcB3-8pB2U2G175td1Hz1RJ/view?usp=share_link)

sean1983, 2023-02-13:
@NTLS09 You should have ALL the colours,

Here's 499,  still missing o[Colors in Sorting](https://docs.google.com/spreadsheets/d/1-mx6Eh0gbIuU5EfUQ4qsPUlTB46uJ8jd7obSIRS_2n4/edit?usp=drivesdk) ver 16millions tho.

NTLS09, 2023-02-16:
Thanks, I'll check it out. I've made the remote for android with the irplus
app which I've posted as a comment. Feel free to try it out.
I'm assuming with the 1s and 0s that they are the actual ir code to use?
I'll batch import them, I'll need to seperate the name with the code with a
comma.

sean1983, 2023-02-16:
Yes,  the 1's and 0's are the very same as ones found in 'effect_definitions.py'.

I will have to get it and have a play.   Interestingly enough I read about IR Plus and it's a software version of the classic Philips Pronto remote.  Which I have....

It actully mentions the compatibility with this classic Pronto,  so I wonder if I could load the data into my physical remote the same way. Rather than the hard way I did,  recording them into remote one at a time.

In this video you can see me controlling them,  but I programmed this remote from Dani's Arduino Sender Script via IR recording them 1 by 1 in to the the arse of the remote.

[image]

This was taken after a Coldplay Gig last year in Glasgow.

I had a bit of fun with remote before gig started too,  also more fun after leaving the venue lighting up crowds leaving and walking home or trying to find taxis.

sean1983, 2023-02-16:
> Thanks, I'll check it out. I've made the remote for android with the irplus
> app which I've posted as a comment. Feel free to try it out.
> I'm assuming with the 1s and 0s that they are the actual ir code to use?
> I'll batch import them, I'll need to seperate the name with the code with a
> comma.

This might same you some time,

CSV Version,  the last 3 columns are the RGB Values  (0-255) of the output colour,  but the names aint acurate after stopping are starting brute force 500 timea you start to loose the plot and see the wrong colours etc, 

You could get rid of them columns in excel or maybe use them to see button colour in IRPlus.

https://docs.google.com/spreadsheets/d/1Q4R1BnwblBfUeCjK12pxlU19X2jkPqj1qeZ7bJfaYVI/edit?usp=drivesdk

Recapi, 2023-03-13:
https://user-images.githubusercontent.com/507161/224833890-50b57d09-a54a-47ff-be77-e22ecea3b08a.mp4

[image]

I'm From Brazil , and Coldplay is playing here for 6 days. and works flawless

NTLS09, 2023-03-26:
Great to hear it works for you @Recapi

NTLS09, 2023-03-26:
Great to hear this works @Recapi

Danielasba, 2024-09-22:
> WhatsApp.Video.2023-03-13.at.6.11.26.PM.1.mp4 
> [image]
> 
> I'm From Brazil , and Coldplay is playing here for 6 days. and works flawless

Please how?