danielweidman/pixmob-ir-reverse-engineering #14: PixMob events thread (post here if you got a wristband somewhere!)

danielweidman, 2023-01-27:
Hello,

I've discovered it's pretty hard to tell where PixMob products have been used in the past/tours where they are currently being used/upcoming events they might be at.

I wonder if we could crowdsource some of this info, so I'm starting a thread here. 

I'll start with some I know of:
- Coldplay "Music of the Spheres" tour (2022)
- Taylor Swift "Reputation" Tour (2018)
- Bad Bunny "World's Hottest" tour (2022)
- Cleveland Cavaliers 2022 Season Home Opener
- Shawn Mendes "The Tour" tour (2019)
- Superbowl (multiple, most recently 2022)
- The Weeknd "After Hours Til Dawn" tour
- Capital FM Jingle Ball (2022)
- Lady Gaga "Chromatica Ball" tour (2022)
- League of Legends World Finals (2022)
- San Jose Sharks (Opening Night 2021, recurring use of PixMob Nova during season?)
- Imagine Dragons "Mercury" tour (2022, only bowl-style venues, not ampitheater-style ones?)
- Elton John 2022 tour (not sure if every show or only some; would love some insight on this)

There are some more listed at: https://www.pixmob.com/projects, however that site doesn't have anywhere near every event on it and doesn't list many current tours.

If you get a bracelet for an event that's not been said already, please leave a comment to let us know and describe the bracelet/other item you got!

tnikitin, 2023-01-31:
I received a PixMob bracelet at the Handball World Championships in Stockholm, Sweden (January 2023). 

Opened it up, it seems like it's using something else than IR. It doesn't have an IR diode, and it looks like it has an rf antenna on one of the layers of the PCB.

PCB contains the following text:
Cement v1.1
20220228

[image]

danielweidman, 2023-01-31:
> I received a PixMob bracelet at the Handball World Championships in Stockholm, Sweden (January 2023).
> 
> Opened it up, it seems like it's using something else than IR as it doesn't have an IR diode.
> 
> PCB contains the following text: Cement v1.1 20220228

Woah, that's interesting. Thanks for the photograph. Perhaps it is a Bluetooth one? https://github.com/MustBeArt/PIXMOB-reversing.

tnikitin, 2023-02-02:
> Woah, that's interesting. Thanks for the photograph. Perhaps it is a Bluetooth one? https://github.com/MustBeArt/PIXMOB-reversing.

The event had 20k+ people, not sure if bluetooth would work with that many people.

> 
> Can you make out the text on the silver component in the top-middle? (and the others too, I guess)

Markings on the PCB
X1:
26.1522
MHZ-H

U1:
CMT2210LH

U2:
AKICW

U3:
No visible text

danielweidman, 2023-02-03:
> Markings on the PCB X1: 26.1522 MHZ-H
> 
> U1: CMT2210LH
> 
> U2: AKICW
> 
> U3: No visible text
> 
> "CMT2210LH is a low power, high performance OOK RF receiver. It is suitable for ISM band 315MHz/433.92MHz wireless applications."
> 
> I've checked the datasheet (https://www.hoperf.com/data/upload/portal/20220218/CMT2210LH%20Datasheet%20EN-V0.6-211214.pdf), with the 26.1522 mhz crystal the frequency would be 915mhz.

Okay, wow! Thanks for thoroughly going through those markings and finding the datasheet. Look like they really are doing a non-BLE RF model of bracelet now! I wonder if they are going to abandon the IR ones completely (TBH that would make me kinda sad) or if this is just a new, different sort of offering. I suppose this would be harder to do location-based targeting with, but they don't always do that with the IR ones anyway.

I looked for recent FCC filings by PixMob, and sure enough I found one for a new 915 MHz OOK transmitter.
https://fcc.report/FCC-ID/2ADS4-WAVENODE/

Looks to support 31 individual bracelet groups. It also mentions and "Impact Mode", like on the older IR bracelets? Did they do any motion-sensitive effects at the event?

ivlzme, 2023-04-03:
I got a bracelet recently at Taylor Swift "Eras" Tour (2023). PCB:
[image]

There is text `VIC v2.3r1 (C) 20211206` likely indicating the firmware version. Looks to be a couple years old. Unlike the above, this uses IR.

You can't make it out on the picture but the component labeled `U2` also has the marking `AKICW`

danielweidman, 2023-04-03:
Thanks very much for the detailed info! I hope you had a good time at the concert :)

adi6409, 2023-06-10:
Looks like the exact same one a friend gave me from the music of the spheres tour!

settwi, 2023-07-08:
i have two PCBs from the Eras tour (June 2023) that are also v2.3r1. I emailed Pixmob to see if they could release their instruction set and didn't hear back lol.

It seems like these bracelets are pre-programmed to shut off a while after the show, even with charged batteries. They don't respond to the demo programs.

[image]

Is there a reset pin on the little 8-pin microcontroller? I might try to figure out a firmware dump...

WS

TheReelDill, 2023-07-10:
Another OOK rf device.  This one from The Weeknd at the London Stadium 
[image]

antonialoytorrens, 2023-07-20:
> 

1A5CEBAA-3F1F-4AFA-A66D-30063EEBB542
Another OOK rf device. This one from The Weeknd at the London Stadium
Uploading BB562A2A-DAB5-4D6E-B263-707D6D8FFA89.jpeg…

This is exactly the same one as The Weeknd at Estadi Olímpic de Montjuïc I had today.

tiix95, 2023-12-14:
> I received a PixMob bracelet at the Handball World Championships in Stockholm, Sweden (January 2023).
> 
> Opened it up, it seems like it's using something else than IR as it doesn't have an IR diode.
> 
> PCB contains the following text: Cement v1.1 20220228
> 
> [image]

Got the same at Amsterdam Music Festival in October. RF one, working well with captures in rf/ folder !

Bardymo, 2024-02-12:
Hello, I was doing a little research on how PixMob bracelets work and on the codes that Danielweidman has given, and I find it interesting. I recently went to Juan Luis Guerra's concert, and they gave us the PixMob bracelets, of course out of curiosity before starting the concert I opened it, and I could see that they use an IR diode accompanied (according to what I was able to investigate) with an nRF52832. Power is provided by two AAA batteries and the bracelet has 4 RGB diodes.
At the concert it was wonderful to see the spectacle of these colorful bracelets, I also don't know how they could create wave-like sweeps and also how they could illuminate one area and not another (I suppose they achieved this effect using projectors with moving heads but I'm not sure) . I tried to find out if there were IR projectors but my cell phone camera is not capable of showing the infrared spectrum.

After the concert they left us the bracelets, and I would like to be able to light them and make some color combinations as I have seen in some short videos. It would be interesting to use those bracelets again in some family activity that one does. I am not an expert in electronics or lighting, but if you could help me turn them on I would be very happy. With the little knowledge I have and using AI, I have tried to use the codes that several people have taught to turn on the lights, but I don't know why they have failed and when I try to execute them it doesn't happen, even when they are compiled on the Arduino. . Here I leave you a photo of the set.
[image]
[image]

AleX04Nov, 2024-05-12:
Got my from Eurovision 2024:
[image]
Looks like it has 24.something crystal

adriah, 2024-05-21:
Got this one at a Taylor Swift Eras concert in Stockholm last Sunday (2024-05-19)

[image]

danielsteiner, 2024-08-23:
Since Taylor Swofts concerts were cancelled in Vienna I‘ve only got some from the Coldplay music of the spheres tour, they are the same as already posted. The Flipper Zero IR files work fine on them!

Hurka007, 2024-11-19:
Cement v1.1 was used on KYGO WORLD TOUR @ Budapest 17th-11-2024.
Is there any way to make it work? It does not have an IR module.

dev464, 2025-06-05:
Shakira concert in Detroit, MI, May 22nd, 2025

[image]
[image]
[image]
[image]
[image]
[image]

InterstellarScout, 2025-06-11:
[image]
This cement v1.1r3 appears to be the latest board. I got it last night from a Weeknd concert. Likely IR, though I haven’t gotten it to respond to anything yet.

diegorocha, 2025-09-09:
NFL São Paulo Game 2025

[image]

kstype-blip, 2025-12-10:
Hi,

I am new here and was hunting around to learn how the PIXMOB wrist devices may work and stumbled into this discussion.

We just attended the Bryan Adams “Roll with the Punches Tour” in October 2025 and accidentally took it home with me.  It’s a very simple package with 4 LEDs:

[image]