danielweidman/pixmob-ir-reverse-engineering #15: Raw Codes For Manual "sendRaw" Arduino Sender

BjarneJensen, 2023-02-18:
Anyone know how to edit the csv generator script to generate raw codes like:
 `"RED": [1462, 1338,  754, 656,  754, 1350,  756, 2748,  754, 2050,  1458, 646,  752, 652,  758, 1346,  1454, 2750,  1464, 2740,  752, 1352,  750, 1350,  1456, 646,  758, 648,  756, 646,  754, 2750,  1456, 2748,  762],
` 
for use with arduino ir sendRaw? 

I got this code from IrRecvDump reading, while running single effect script to esp32 serial. The code works, but its a roundabout way to get the values I'm looking for. I can't wrap my head around how get these values by operating on the entry
`"RED": [1, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 0, 0, 0, 1, 0, 0, 0, 1, 1, 0, 1, 0, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 1, 0, 0, 0, 0, 1],` in the definitions file with the csv generator script and would rather not re-decode them to make a list of raw commands. I probably missed something in the docs. 

Thanks for any help you can offer.

danielweidman, 2023-02-20:
Hello!

Each 1 or 0 in the arrays in `effect_definitions.py` represent a 700-microsecond* period of time, during which the IR signal was either "on" or "off".

In the "raw" signals, like the one you got from IrRecvDump or will need to send with sendRaw, the numbers represent the # of microseconds between changes of the IR signal between "on" and "off".

As an example, let's look at the first few bits (1/0s) in the "RED" signal:
`"RED": [1, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 0, 0, 0, 1, 0, 0, 0, 1, 1, 0, 1, 0, 1, 0, 0, 1, 1, 0, 0, 0, 0, 1, 1, 0, 0, 0, 0, 1],`
We get: 2 "on", then 2 "off", then 1 "on", then 1 "off", then 1 "on", then 2 "off", etc.

- 2 "on" = 2*700μs = 1400μs
- 2 "off" = 2*700μs = 1400μs
- 1 "on" = 1*700μs = 700μs
- 1 "off" = 1*700μs = 700μs
- 1 "on" = 1*700μs= 700μs
- 2 "off" = 2*700μs = 1400μs

If we want the representation like that used by IrRecvDump or sendRaw, we need to just put together the multiplied out time intervals. So, for "RED", it starts `[1400, 1400, 700, 700, 700, 1400...`. This is what you captured with IrRecvDump.

If you want to perform this calculation programmatically, you can use [`to_data_timings()`](https://github.com/danielweidman/pixmob-ir-reverse-engineering/blob/879bfc9d9e5a17645b947ed48b7e825a1ae08f84/python_host/pixmob_conversion_funcs.py#L38) from `pixmob_conversion_funcs.py`. That function will take in an array of 1s and 0s and give you back an array of the transition timings.

Let me know if this all makes sense or not.

_*When you manually capture with IrRecvDump, the numbers won't be exact multiples of 700 due to noise, but they are pretty close._

BjarneJensen, 2023-02-21:
Thanks so much for the speedy response and in-depth explanation. It makes total sense now. [to_data_timings()](https://github.com/danielweidman/pixmob-ir-reverse-engineering/blob/879bfc9d9e5a17645b947ed48b7e825a1ae08f84/python_host/pixmob_conversion_funcs.py#L38) is exactly what I was looking for! Thanks for your work to make this possible. 

I work near a venue that uses these bracelets. Hoping to get some raw data to contribute soon. I'm most interested in the "stay on" command, as I have seen people leaving the venue with their bracelets still lit up out of range of the ir emitter. Also interested in dmx to ir.

closed 2023-02-21 by BjarneJensen

danielweidman, 2023-02-21:
@BjarneJensen Great! I'm glad that function will work for you.

I'm also very interested in the "stay on" command--getting raw data for that would be excellent :)

Also, if you see specific events that the bracelets are used in, feel free to add them to [this thread.](https://github.com/danielweidman/pixmob-ir-reverse-engineering/issues/14)

Thanks!