danielweidman/pixmob-ir-reverse-engineering #16: Somethings Interesting... EEPROM

sean1983, 2023-02-20:
I noticed on board there wristbands there is something attached the the MCU (Processor) that looks like a Voltage Regulator,  Turns out its not... It's a small EEPROM.

It's a 'Microchip 24C02' 5-Wire Version.

[Microchip 24C02 Datasheet](http://doc.chipfind.ru/microchip/24c02.htm)

So i've removed the IC from a sacrificial PCB and build a little reader up on an Arduino Uno.

I have no idea is this can be descrambled, decoded or converted to anything more useful.

But here is the Hex Dump!

`
B5 CD 61 E9 DE DF EB FC 3F EE 3F 4D E0 BF C7 F3 
93 77 8D EB CE FD B6 2B 8F F3 4E DB DD 96 F7 AD 
FD 67 7C FC F3 43 9F BF 7B 57 F7 FB BD EC 53 DC 
DB B7 29 F8 ED 77 CF F1 AF DC 7E F4 7D 3F 9E 7E 
DF 74 A3 E6 D2 7D 55 DD 7E BB 76 6E B5 7F E8 27 
8E 7A 7F EB 81 F9 E7 76 A6 B7 01 FF 05 9F 77 BE 
9E C9 3D DB FD FB 2E 6B F5 7F F5 EF EF 74 FD 23 
FE D5 47 3B AF F3 F6 AB 7B 76 43 71 E6 59 95 7D 
E3 93 7F F1 BA F6 DF FF F5 34 65 FB CD ED 56 5E 
37 DF E5 ED D5 0D 9E 7B 7B A6 53 DD 47 B3 F7 FB 
DE F7 37 94 C4 2E 43 DB E6 A7 6B B6 F6 FD AF D4 
EB BF 7A 59 97 7B FF 4F AF F2 F6 7B EB 1D 77 79 
FF 1D B6 3B FB 58 EF B3 B1 86 B5 DB E3 3B F7 7E 
76 3D CF 78 FE B1 53 FD F9 B5 E7 F3 76 5F 23 B7 
7D A4 95 BD 52 EF 7E 57 AF 9F 6D B5 FF DF 7D 8F 
2E 3F 9F 8E FF 9F E3 A7 6E 4D 65 17 7F 7F 3F CF 
DF 8A 74 BF 7D BA 66 7F FF F3 FE AE 83 D7 67 53 
A1 A3 67 3F E1 A7 9E E0 DD FF F1 D7 47 6B 75 F7 
D9 F9 56 D7 FF FD FA 69 FF 94 B7 FF DF 3E FB FD 
FF E3 5E ED 7A 37 0F D2 B9 3A BF FC F3 3F A4 EC 
D5 FE 6E F9 67 FB 5B 7C 0F 3F F4 C7 C7 AF 6D 4A 
F3 36 1F 7F F9 FB 7E AA F9 C7 D9 DF 5C 3E 3A DD 
FF E2 C1 EC 91 DE BD 5D F7 9F BD 7F 5B B7 FB B9 
AF D9 9F AF BB 79 EE FB EB 6B DC CF D9 F8 93 0C 
FC FE 7E EF 34 F6 B7 EF F5 4D 7F D9 7E 57 CF FA 
DF A4 F6 35 DD 24 B8 7A 2B F7 7F E3 73 BE CE FF 
E7 8D 8E 55 9B 8A CF FD E7 33 FF F7 2E B8 7D BD 
FF AF FB F5 AC F9 92 FE 3B 93 7D 8D 2F 77 A7 D6 
EF 71 FF D5 75 EB F7 58 E5 EF 83 FB C7 E1 F7 1E 
E4 ED 09 EB DF F4 E7 FF EC FC F6 69 DC 6F B6 FB 
C6 94 F3 A8 67 A4 A7 78 B3 C9 7F FB 1D E5 FF C4 
17 FA FD 51 FB 3E 1D CB EB DF 5B FC BF CE E2 DB 
3F FD A0 3E AF 79 E9 F4 6B FE F7 6F ED ED D2 D1 
D7 99 F2 DB 7F FB BE E8 FA DB DF F2 BD 7E AB EB 
DA B3 E0 3F 69 FD 64 2E 7E 55 DA 2F 96 66 B7 ED 
CF B4 1D 83 5D E1 BF DA A5 BF 91 AF F3 69 8F 75 
6F 1E BA FB DF DE EB 77 E9 75 78 6B 85 5A 0F 5B 
DD F2 CA 7A D5 BB CE EE A5 13 EB BF BF 3F DD 1B 
B0 3F 69 FD 5B E7 E6 FF E6 BB EB F6 DF 82 8B 1F 
7E AD 6F DE 7F F2 CD 1B DE FC D5 AD BF D7 AC F5 
9D 73 FD FB FF CF 6F B7 FB EF 0F 7B CB 64 BD FB 
AF F5 1F CB F8 17 65 96 36 97 23 5F C4 FD E3 D7 
57 5E FB 6D 34 2B FC 6E ED 0F A5 2E BE FE 99 8C 
FF BF 11 ED 5D B5 FA 9D FE 75 F6 BD EB 5F FF FA 
8E F8 B3 ED ED BE FB CB E7 AD 3D D1 BE 91 5C 3F 
5F BE 2A F9 C5 C7 DF 9F EF 04 EA 01 95 01 FE 19 
1A 01 0A 00 02 B5 48 23 A3 00 01 04 EA 04 EA 01 
C1 07 3B 07 4E 0E 9C B5 FF 00 A3 00 00 EC 04 5C 
FC CB F5 71 F6 E7 6D 6B C8 F1 56 EC D7 D6 85 FF 
EF FC EA 7C 59 1F FB DF F3 99 58 D0 F9 B3 5D 4F 
FD EB C9 5D 43 C3 DE 83 EF F9 4D DF 3F 73 3B A0 
7F 83 6B F9 B9 FD B9 DC F5 0D 5F E5 71 FB DF DA 
FF AB FF C1 94 FB 3E E3 BA E2 FB DB EF F7 DB 25 
B7 BE FD EB EF F4 BD 5F E2 8F FE D7 57 ED FF EC 
7D 17 E5 FD F7 6C D3 98 DB 1B ED 6F BE 3F DB DE 
FF 2E AE 3E DC 7A 7D E7 FE 69 3F 19 BE 7B B8 89 
FD F9 5C 41 9E 8A 75 CF 59 AD F5 ED E5 6E EF 45 
C0 8B F1 BE E7 6F CE FF 33 7E 9E F9 1F FD D2 ED 
5F 09 9E 1A FF C3 CF 0F E5 EA EF E3 BC 7C 1E FF 
3E FC F7 6F 7E 2E 7B 66 BC F1 57 E7 FF DF 7F FD 
DF 5D 76 9F 2F B5 57 FA FF 4A 6A C6 2F AA E5 ED 
CC DD B7 F0 BE AE 7F BB BE 53 FF B9 7C 03 D9 01 
95 02 04 3A 3B 08 34 00 03 00 00 23 A3 00 01 C7 
08 F3 08 F3 08 F0 01 14 08 D2 08 F0 00 00 00 13 
`

danielweidman, 2023-02-20:
Wow, thanks for sharing this dump.

I tried looking at it with a few different character encodings and (unsurprisingly) nothing jumped out immediately. I also tried applying [binwalk ](https://github.com/ReFirmLabs/binwalk), and it didn't find anything.

There has be some useful data in here though somehow!

sean1983, 2023-02-20:
It will no doubt be the firmware settings stored in this EEPROM,  With the chip being Read/Write it makes sence,  As i think the ABOV MCU they use is only Single Shot Write,   So this EEPROM is probably the only difference betweeen the IR, RF, BLE and Vibration models and the MCU is likely the same on all of them.

I have a feeling its encrypted,  I would guess AES Encrypted.  So without the Private key,  Likely useless.  But who know's.... Someone might see or know more!

sean1983, 2023-03-26:
Added the EEPROM Dump to my fork,  And in its own folder inside of MISC,  I will close this this Issues now!

closed 2023-03-26 by sean1983

Luca-Ricci, 2023-08-19:
Hi, I was doing a bit of study on bracelets. Your work could be really useful to me! I wanted to know, in addition to the ".hex" file, were you able to obtain a ".eep" file or equivalent? I would like to see what is in the eeprom registers