danielweidman/pixmob-ir-reverse-engineering #24: Pixmob Bracelet Wave Radio Frequency Hack.

ericmargay, 2023-04-19:
In the recent concerts they are using this kind of bracelet. I am searching to the web if there is any hack. Apparently doesn't exist info about it. Do you think IR bracelet has the same chip as the RF bracelet ?.

Adding some pictures.
[image]
[image]
[image]

sean1983, 2023-04-19:
Hi, Thanks for your post, We are in the process of trying to decode them,  we had only had them a few days,   I have done my research on the main MCU IC and it looks to be the same,  Also the new addition Radio / OOK Decoder IC is interesting and the datasheet states it to be a direct replacement for Infrared,  So I am betting that IC sends the same old codes to the MCU after being demodulated/decoded from the OOK Modulation that is used via RF.  

But if you have a Flipper or HackRF, or DIY 915mhz  ardunio TX, Then we already have a few codes that will work.

ericmargay, 2023-04-20:
> @ericmargay What concert did you get this from, by the way? I can share recordings of the signals soon

In Mexico City a mexican musical named Grupo Firme, in Foro Sol, February this year.

ericmargay, 2023-04-20:
> @ericmargay What concert did you get this from, by the way? I can share recordings of the signals soon

I can share also in the next concerts.

ericmargay, 2023-04-20:
> Hi, Thanks for your post, We are in the process of trying to decode them, we had only had them a few days, I have done my research on the main MCU IC and it looks to be the same, Also the new [...]

I will get a HackRF to use the codes. Contact me to follow-up.

TieHaxJan, 2023-04-30:
Hey guys, 
from 25th of April to 28th of April was the ProLight&Sound Fair in Frankfurt, Germany, Pixmob was there and I've got 9 of the Radio Frequency Controlled Bracelets, if I can help in any way, just contact me, I also can get ahold of a HackRF from my University. Sadly I was not able to bring any device to record their signals and also on the fair there was way to much radio noise, I would assume.
Cheers.

dontic, 2023-05-02:
> Hi, Thanks for your post, We are in the process of trying to decode them, we had only had them a few days, I have done my research on the main MCU IC and it looks to be the same, Also the new [...]

@sean1983 I have one of these Bracelets from a Swedish House Mafia concert and a Flipper, sadly I did not have it then so I do not have any recordings. Let me know if I can help in any way.

dontic, 2023-05-03:
@sean1983 yes, pity I did not tinker with it sooner. This concert was in Madrid, October 14 2022. The venue was a new big dedicated hall in IFEMA, Madrid's fair grounds. Photo attached.

In [PIXMOB's site](https://www.pixmob.com/products/detail/led-wristbands) if you head to the Waveband  section it actually states "As seen on Swedish House Mafia and more.".

Will gladly join the discord, see you all there.
[image]

below, 2023-07-15:
[image]
[image]

Is this the device we are talking about (The Weeknd: After Hours Till Dawn)? I have two of them, both not deactivated. My questions:

* Can they be opened (relatively) non-destructively?
* What do I need to make them blink?
* Can I help anyone by donating my second device?