danielweidman/pixmob-ir-reverse-engineering #36: issue with running files

Capester1, 2023-06-11:
Hello,

I am trying to use an Intel MacBook Air to run the python scripts.

I was wondering how you are supposed to download the files and then run them to light up the pixmob bracelet. I currently tried downloading the config.py file and the demo_single_effect.py files to a folder on my desktop. Are there others that I need to download? How do you run the files? Any help is appreciated.

Thanks in advance!

sean1983, 2023-06-11:
Hi, 

Yes there is other files You  should see them in Imports at the begining of each python script,

- pixmob_conversion_func.py
- effect_definitions.py

Also make sure that the modules are installed too, like PySerial using PIP INSTALL

Be sure your directory structure is correct matching the repo.

You would probably be best cloning or downloading all the files from the Repo.

Check your Serial Port and Baud Rate to defined correctly in config.py

Hope this helps 

Sean

Capester1, 2023-06-12:
Thank you. It says that it has successfully sent the code to the arduino. However, I am testing with a standard visible LED, and I am not getting it to light up. however, the 'L' LED on the arduino UNO circuit board is illuminating briefly when I run the demo code. Is there something that I should change in the arduino code? I already have it assigned to what pin the LED circuit is connected to. Thanks.

Capester1, 2023-06-13:
Hi,

It appears to be a PixMob X3, according to pictures on the website. It is from the Taylor swift 2023 concerts. 

Is there anything else other than the data pin that is being used that needs to be entered into the PixMob_Transmitter_Arduino.ino file? 

Thanks.

sean1983, 2023-06-14:
Hi again,

I guess it's tbis model you mean that Talor Swift uses.

[image]

[image]

It might be worth checking the batteries,  These don't hols charge very long,

With this model they made them NON-Recyclable by heat-staking them together,  but you can snip the snip the back of the melted plastic stakes and seperate it to test or  replace the batteries.  If cut flush it clips back together again.

[image]

[image]

[image]

With the Arduino script it is just the GPIO Pin that needs to be specified.

May I ask what type of Arduino you are using ?

Capester1, 2023-06-17:
Hello, Thank you for all of your help. I was able to get it to light up by setting `WAIT_BEFORE_SEND = True ` in Config.py.

closed 2023-06-17 by Capester1