danielweidman/pixmob-ir-reverse-engineering #37: loop effect continuously

whal3y, 2023-06-17:
Hello!

I received a PixMob bracelet at a Taylor Swift concert in Chicago, Illinois. To my suprise, after the show, the bracelet retained an effect, and it is now looping a pink fade in-out effect. I set up and successfully executed the code on one band I had laying around, but I have intentionally not connected the one that currently lights up. Is there a way I can setup a loop similar to this, in a way that it does not need to be in constant range of the IR LED?

Thank you for your insights.

danielweidman, 2023-06-17:
Hello!

We have a couple different loop effect codes through brute force and
recording analysis, though we haven't figured out how to make arbitrary new
loop effects.

Try these and let me know if they work for you:

1,1,0,0,0,1,0,0,1,0,1,0,1,0,1,1,0,1,0,0,0,0,1,0,0,1,0,0,0,0,1,0,0,1,0,0,1,0,1,1,0,0,1,1,0,1,1,0,0,0,1,1,0,0,1,0,0,1,0,0,0,0,1

1,1,0,0,0,1,0,0,1,0,1,0,1,0,1,1,0,0,0,1,0,0,1,0,0,1,0,0,0,0,1,0,0,1,0,0,1,0,1,1,0,0,0,1,0,1,1,0,0,0,0,1,0,0,1,0,0,1,0,0,0,0,1

1,1,0,0,0,1,0,0,1,0,1,0,1,0,1,1,0,0,1,0,0,0,1,0,0,1,0,0,0,0,1,0,0,0,1,0,1,0,1,1,0,0,0,1,0,1,1,0,0,0,1,1,0,0,1,0,0,1,0,0,0,0,1

All the best,

Dani

whal3y, 2023-06-18:
Thank you, these effects worked nicely. Is there any way that you can make it fade between colours? if not, no worries.

sean1983, 2023-06-18:
Hi,  Some colours that stay light  for a while or with a slow fade will actually fade to another colour,  but only if the second code is sent while the previous is still active but only some do this, 

So you would have to experiment to find which ones it will work with.

We haven't actually logged which colour codes have that behaviour, but something we should keep a log off actually!

whal3y, 2023-06-23:
hello again, I have tried that and I have found a few that fade into each other. 

by any chance, are there any more loops that have been discovered? I have started decoding the data and through brute force I have found another blue variant:

1, 1, 0, 0, 0, 1, 0, 0, 1, 0, 1, 0, 1, 0, 1, 1, 0, 0, 1, 0, 0, 0, 1, 0, 0, 1, 0, 0, 0, 0, 1, 0, 0, 1, 0, 0, 1, 0, 1, 1, 0, 0, 0, 1, 0, 1, 1, 0, 0, 0, 1, 1, 0, 0, 1, 0, 0, 1, 0, 0, 0, 0, 1 

it produces a blue-green colour and I think it just blended the 2 I was trying to compare. by any chance has a red or yellow loop effect been discovered? I want to test my theory to see if I can try and decode the formula.

theimpossibleheir, 2024-08-07:
Hello. I just came across with shmmr in this website. https://beta.shmmr.app/register

I made my own shmmr device following all the instructions. 
[image]
[image]
[image]
[image]
[image]

uploading this arduino "into" file to the board. it does work
[ShmmrTransmitterFirmware_v1_2 (1).zip](https://github.com/user-attachments/files/16516640/ShmmrTransmitterFirmware_v1_2.1.zip)

MY PROBLEM IS, IT ONLY FLASHES ONE COLOR PERSISTENTLY LIKE THE COLOR BLUE OR GREEN OR RED OR WHITE. 

Can someone make a revision to the shmmr arduino "into" file to have loop effect continuely with all the colors since i don't really know how to do it? 
[image]
[image]

https://github.com/user-attachments/assets/8d08a191-d264-4730-9d49-065e251b8d8b

https://github.com/user-attachments/assets/7b636505-fea5-4c1c-bb41-bed60ec1956b

https://github.com/user-attachments/assets/05885455-22e2-467f-af74-6d7cac15767b

[image]
[image]

That's all the reference. This shmmr file does work. but i need someone to revise it for all colors like red, fade red, green, fade green, light green, fade light green, yellow fade, blue, blue fade, light blue, light blue fade, magenta fade, pink, white, tuquiose, and so on. 
[ShmmrTransmitterFirmware_v1_2 (1).zip](https://github.com/user-attachments/files/16516703/ShmmrTransmitterFirmware_v1_2.1.zip)

https://beta.shmmr.app/register

thank you very much