danielweidman/pixmob-ir-reverse-engineering #38: RPi IR sending

settwi, 2023-06-25:
Heya,

I'm trying to get some transmitting code up and running using the hardware PWM on an RPi Zero W for these Pixmob bracelets.

I've seen some stuff online for IR control using the [`lirc` library](https://www.lirc.org/), but the issue with this library is it requires you to press buttons to register the corresponding "remote."

I think an easier and more customizable way to do this would be to just use the [`bcm2835` PWM API](https://www.airspayce.com/mikem/bcm2835/) directly. You can set a 38 kHz frequency and then "bit-bang" the delays corresponding to 1's and 0's specified in the Pixmob communication protocol that you all have been decoding.

I'm imagining I can just take the binary codes that you send to the Arduino serial port and send em off using the PWM idea once the delays are properly incorporated. I'll try to make a simple C program that you could call from subprocess in Python, or maybe just make a Python/C bridge so you don't need to use subprocess.

This is in its infancy but I wanted to just put some info up here and get y'alls thoughts on this.

Thanks!
best
William

settwi, 2023-06-25:
One thing I'm trying to figure out is the "clock divider" that we need to set in order to get a 38 kHz square wave for PWM.
I think we can interpolate a function based on [the given divisors](https://github.com/janne/bcm2835/blob/88812525bb36ee27d350502d86847ac8c311067f/bcm2835.h#L1145C1-L1159C26) to figure that out.

settwi, 2023-07-08:
@sean1983 or @danielweidman could you try out this repo i've been working on (using a Raspberry Pi) and see if you can get some bracelets working? i think the ones i have from a Taylor Swift concert are programmed to deactivate.

https://github.com/settwi/infrared-kernel-blaster

The repo is only compatible with the BCM2835 chip...

You have to `make` the kernel module and then do `make test` to insert it into the kernel. Then you can `echo aabfccedff12345 > /dev/blaster` where that string after `echo` is the hex code of the command to send.

Connect the LED driver circuit to GPIO pin 12.

You might have to install the RPi kernel headers for this to work... I will update the readme soon.

Best
WS

avolk201, 2024-03-04:
@settwi I know it's been a while since you posted this, but what did you mean by "i think the ones i have from a Taylor Swift concert are programmed to deactivate." Do they not flash after the fact? Or are they just uncontrollable? I'm thinking about attempting this once some components arrive tomorrow but I only have the ones from the Eras Tour.

settwi, 2024-03-04:
It was user error on my part. Code works!

closed 2024-03-04 by settwi