danielweidman/pixmob-ir-reverse-engineering #39: Music Of The Spheres Coldplay Wristband

pochopsp, 2023-06-29:
Hello all, nice to meet you!

I'm an Italian software engineer, and I happen to be a Coldplay fan.
I went to their concert in Naples some days ago and I brought the wristband home with me (see attached pic), thinking that I could find some way to turn it on again.

[image]

After a little searching online, I stumbled upon your github repo.

Since I don't know anything about these wristbands, arduino, IR or RF, could you please help me in turning it on?
What do I need to do to achieve this (is there something I have to buy? is disassembly needed?)

Many thanks for your help.

Best regards,
Pasquale

P.S.
I know this isn't much of an issue, I'm sorry for that. Please feel free to close it if you want, and if you can we can continue our discussion via email, my address is pasquale.1996@gmail.com

pochopsp, 2023-07-07:
Well, since I've already bought IR transmitter and arduino online, I really hope it's infrared.

But as far as I know, all wristbands used in coldplay music of the spheres were infrared...

Or am I wrong?

settwi, 2023-07-07:
It looks like IR based on the little dot on the front, but I think some concerts over in Europe are using the RF modules.

As for powering the wrist strap back on, you probably need to disassemble it and get new batteries. It takes two little 3V coin cells in series

pochopsp, 2023-07-07:
Actually it has standard AAA batteries 
[image]

sean1983, 2023-07-07:
Sorry for the late reply guys,  I've been recovering from Glastonbury Festival.

All the Coldplay ones are Infrared.

You basicly just need an 840nm IR LED and an Arduino and you can use the code on the Repo to  get them working

pochopsp, 2023-07-07:
@sean1983 wow! Glastonbury must have been awesome!

I've bought this transmitter:
https://it.aliexpress.com/item/1005004005620672.html?spm=a2g0o.productlist.main.1.46d114ca9nZ5qn&algo_pvid=2a039d4b-4579-4bd2-a558-635ca88bfee2&aem_p4p_detail=202307020843401456071700085640004513083&algo_exp_id=2a039d4b-4579-4bd2-a558-635ca88bfee2-0&pdp_npi=3%40dis%21EUR%210.38%210.3%21%21%21%21%21%402100bbf516883126200237378d0745%2112000027716275871%21sea%21IT%210&curPageLogUid=JxAVUbP7r5Kf&search_p4p_id=202307020843401456071700085640004513083_1

and this arduino - clone:
https://it.aliexpress.com/item/1005002997846504.html?spm=a2g0o.productlist.main.5.2b31suNLsuNL9R&algo_pvid=0b1e839c-a183-426e-ab8f-efcfa5a6ebb2&aem_p4p_detail=202307020850351221903485349820009559360&algo_exp_id=0b1e839c-a183-426e-ab8f-efcfa5a6ebb2-2&pdp_npi=3%40dis%21EUR%213.45%213.1%21%21%21%21%21%402100bb6416883130350831592d077e%2112000023136335076%21sea%21IT%210&curPageLogUid=PlFscdDwMyvL&search_p4p_id=202307020850351221903485349820009559360_3

I hope they'll be good for the purpose

daniele8808, 2025-05-14:
any news about this?
did you find a way to make it work?

Daniele