danielweidman/pixmob-ir-reverse-engineering #41: Keep leds on after command finishes

enyarendon, 2023-09-07:
Hi, is there a command we can send to keep the leds on with a specific color forever?

BjarneJensen, 2023-09-10:
My solution to get a permanent solid color is to send a solid color command, followed by a permanent fade command: for instance: 

```
solid_green_60:[1, 1, 0, 1, 1, 0, 1, 1, 0, 1, 0, 0, 0, 0, 1, 0, 0, 0, 1, 0, 0, 0, 1, 0, 0, 1, 0, 0, 0, 0, 1, 0, 0, 1, 0, 0, 0, 0, 1, 0, 0, 0, 0, 1, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1, 0, 0, 0, 0, 1]
followed by 
permanent_fade_green:[1, 1, 0, 0, 0, 1, 0, 0, 1, 0, 1, 0, 1, 0, 1, 1, 0, 1, 0, 1, 0, 0, 1, 0, 0, 1, 0, 0, 0, 0, 1, 0, 0, 1, 0, 1, 0, 0, 1, 1, 0, 0, 1, 1, 0, 1, 1, 0, 0, 0, 1, 1, 0, 0, 1, 0, 0, 1, 0, 0, 0, 0, 1]

```
It should then fade from green to green, but the fade isn't really detectable. If you don't mind the fade, the second code should stay on on its own. I dont't have a list of other colors at the moment, but you can find more by loading these into bit flippr or brute force

canonborn, 2023-10-07:
@BjarneJensen this worked great!!! how did you find the respective codes for "solid_green_60" and "permanent_fade_green"? I want now to be able to set every color.

theimpossibleheir, 2024-08-07:
Hello. I just came across with shmmr in this website. https://beta.shmmr.app/register

I made my own shmmr device following all the instructions. 
[image]
[image]
[image]
[image]
[image]

uploading this arduino "into" file to the board. it does work
[ShmmrTransmitterFirmware_v1_2 (1).zip](https://github.com/user-attachments/files/16516640/ShmmrTransmitterFirmware_v1_2.1.zip)

MY PROBLEM IS, IT ONLY FLASHES ONE COLOR PERSISTENTLY LIKE THE COLOR BLUE OR GREEN OR RED OR WHITE. 

Can someone make a revision to the shmmr arduino "into" file to have loop effect continuely with all the colors since i don't really know how to do it? 
[image]
[image]

https://github.com/user-attachments/assets/8d08a191-d264-4730-9d49-065e251b8d8b

https://github.com/user-attachments/assets/7b636505-fea5-4c1c-bb41-bed60ec1956b

https://github.com/user-attachments/assets/05885455-22e2-467f-af74-6d7cac15767b

[image]
[image]

That's all the reference. This shmmr file does work. but i need someone to revise it for all colors like red, fade red, green, fade green, light green, fade light green, yellow fade, blue, blue fade, light blue, light blue fade, magenta fade, pink, white, tuquiose, and so on. 
[ShmmrTransmitterFirmware_v1_2 (1).zip](https://github.com/user-attachments/files/16516703/ShmmrTransmitterFirmware_v1_2.1.zip)

https://beta.shmmr.app/register

thank you very much