danielweidman/pixmob-ir-reverse-engineering #52: Can the LEDs stay on?

fapeg, 2024-08-12:
Hey, I installed the remote on my Flipper and it works perfectly.
But is there a way (yet) where you can send IR codes to the bracelets and they stay in this color until you turn them off or the batteries run out? Would be cool to build a reading lamp or something :)

fapeg, 2024-08-12:
https://github.com/danielweidman/pixmob-ir-reverse-engineering/blob/0c1e3345c4ef7d7452bcaba48eca16a4e5721e1d/python_tools/effect_definitions.py
In this file I found the "always on white" command starting in line 256. Do you think I can use the WEIRD_82 until WEIRD_90 commands without messing up my bracelet? Because there's a warning that these "weird" codes might require your bracelet to be reset.

devinbaeten, 2024-08-24:
I found these persistent effects via brute-force. Note that these are pulsing effects and not solid colors. Also, they will persist even after the band is restarted. But will be cleared when another command is sent.
[ir_results_20240824_003610.json](https://github.com/user-attachments/files/16735813/ir_results_20240824_003610.json)

fapeg, 2024-08-25:
Thanks! I converted them for the Flipper, if someone else is looking for Flipper loop files, I'll upload them here.

```
Filetype: IR signals file
Version: 1
# 
name: BLUE
type: raw
frequency: 38000
duty_cycle: 0.330000
data: 1388 2082 694 1388 694 694 694 694 694 694 1388 694 694 2776 694 1388 694 2776 694 1388 694 1388 694 694 1388 1388 1388 694 1388 2082 1388 1388 694 1388 694 2776 694
# 
name: BLUE_2
type: raw
frequency: 38000
duty_cycle: 0.330000
data: 1388 2082 694 1388 694 694 694 694 694 694 1388 2082 694 1388 694 1388 694 2776 694 1388 694 1388 694 694 1388 2082 694 694 1388 2776 694 1388 694 1388 694 2776 694
# 
name: GREEN
type: raw
frequency: 38000
duty_cycle: 0.330000
data: 1388 2082 694 1388 694 694 694 694 694 694 1388 1388 694 2082 694 1388 694 2776 694 2082 694 694 694 694 1388 2082 694 694 1388 2082 1388 1388 694 1388 694 2776 694
# 
name: LIGHTBLUE_SLOW
type: raw
frequency: 38000
duty_cycle: 0.330000
data: 1388 2082 694 1388 694 694 694 694 694 694 1388 694 694 694 694 1388 694 2082 694 2082 694 1388 694 1388 694 694 1388 2082 694 694 1388 2776 694 1388 694 1388 694 2776 694
# 
name: TEAL_FAST
type: raw
frequency: 38000
duty_cycle: 0.330000
data: 1388 2082 694 1388 694 694 694 694 694 694 1388 694 694 694 694 1388 694 1388 694 2776 694 1388 694 1388 694 694 1388 694 694 694 694 694 1388 2776 694 1388 694 1388 694 2776 694
# 
name: PURPLE_FAST
type: raw
frequency: 38000
duty_cycle: 0.330000
data: 1388 2082 694 1388 694 694 694 694 694 694 1388 2082 694 1388 694 2082 694 2082 694 1388 694 1388 694 694 1388 694 694 694 694 694 1388 2776 694 1388 694 1388 694 2776 694
# 
name: BLUE_FAST
type: raw
frequency: 38000
duty_cycle: 0.330000
data: 1388 2082 694 1388 694 694 694 694 694 694 1388 2082 694 1388 694 1388 694 2776 694 1388 694 1388 694 694 1388 694 694 694 694 694 1388 2776 694 1388 694 1388 694 2776 694
# 
name: DARK_PURPLE_FAST
type: raw
frequency: 38000
duty_cycle: 0.330000
data: 1388 2082 694 1388 694 694 694 694 694 694 1388 694 694 2776 694 2082 694 2082 694 1388 694 1388 694 694 1388 694 694 694 694 694 1388 2776 694 1388 694 1388 694 2776 694
# 
name: BLUE_SLOW
type: raw
frequency: 38000
duty_cycle: 0.330000
data: 1388 2082 694 1388 694 694 694 694 694 694 1388 2082 694 1388 694 1388 694 2776 694 1388 694 1388 694 694 1388 2082 694 694 1388 2776 694 1388 694 1388 694 2776 694
# 
name: TURQUOISE_SLOW
type: raw
frequency: 38000
duty_cycle: 0.330000
data: 1388 2082 694 1388 694 694 694 694 694 694 1388 694 694 694 694 1388 694 1388 694 2776 694 1388 694 1388 694 694 1388 2082 694 694 1388 2776 694 1388 694 1388 694 2776 694
# 
name: BLUEISH_WHITE_FAST
type: raw
frequency: 38000
duty_cycle: 0.330000
data: 1388 2082 694 1388 694 694 694 694 694 694 1388 1388 694 2082 694 2082 694 2082 694 1388 694 1388 694 694 1388 694 694 694 694 694 1388 2776 694 1388 694 1388 694 2776 694
# 
name: TURQUOISE_SLOW_2
type: raw
frequency: 38000
duty_cycle: 0.330000
data: 1388 2082 694 1388 694 694 694 694 694 694 1388 1388 1388 1388 694 1388 694 2776 694 1388 694 1388 694 694 1388 2082 694 694 1388 2776 694 1388 694 1388 694 2776 694
# 
name: BLUE_FAST_2
type: raw
frequency: 38000
duty_cycle: 0.330000
data: 1388 2082 694 1388 694 694 694 694 694 694 1388 1388 694 2082 694 1388 694 2776 694 1388 694 1388 694 694 1388 694 694 694 694 694 1388 2776 694 1388 694 1388 694 2776 694
# 
name: PURPLE_SLOW
type: raw
frequency: 38000
duty_cycle: 0.330000
data: 1388 2082 694 1388 694 694 694 694 694 694 1388 2082 694 1388 694 2082 694 2082 694 1388 694 1388 694 694 1388 2082 694 694 1388 2776 694 1388 694 1388 694 2776 694
# 
name: BLUEISH_WHITE_SLOW
type: raw
frequency: 38000
duty_cycle: 0.330000
data: 1388 2082 694 1388 694 694 694 694 694 694 1388 1388 694 2082 694 2082 694 2082 694 1388 694 1388 694 694 1388 2082 694 694 1388 2776 694 1388 694 1388 694 2776 694
```

s00500, 2024-09-02:
Hm I just tried these, seemed to not work, while all other codes are fine, may I ask what band you are using?