danielweidman/pixmob-ir-reverse-engineering #57: Stand-alone controller. Can I help ?

vicnevicne, 2025-03-24:
Hi,

First let me thank you for this amazing project ! 
I think I can help regarding one of the next steps listed in the readme: "Write a standalone Arduino library for controlling the bracelets without a computer."

I got to see a gig of the Indochine (French band) and they use Diamond V4.4A Pixmob wristbands (plus, they also have a matrix of them on the ceiling of the concert rooms) - see for example https://youtu.be/Z9o13vc-MQU?t=121
So, using this project, I first confirmed it worked as expected on the wristband I brought home, then went on and built a stand-alone, battery powered ESP32-based controller that I could sneak into the arena at one of their next gigs.
[image]
(based on https://www.aliexpress.com/item/1005006602400485.html )
I was lucky to see them a second time, and after pranking some people while we were waiting for the show to begin, I tried pointed the device at the ceiling during a pause in a song and well, it works pretty well : https://youtube.com/shorts/D0sr_oAod68
(if Indochine's lighting director reads this: Sorry man, there was no bug in your system ;-))

Long story short, I'd be glad to contribute that Arduino ESP32 code if you're interested. Just tell me and I'll clean it up and wrap up a PR.

bastian123321, 2025-07-17:
My friend, can you send me the ESP32 code?
Best Regards

vicnevicne, 2025-07-19:
Hi,

Sorry for the delay. It's been some time ;-)

First, as requested, here is the final code I used for that prototype:
[PixMob_StandAlone_ESP32_v6.zip](https://github.com/user-attachments/files/21328142/PixMob_StandAlone_ESP32_v6.zip)
By default, the button triggers a "slow white fade in/out", but as we are on an ESP32, the device also opens a Wifi Access Point you can connect to with your phone (set a password in CaptivePortal.ino, line 107). You can then control the ESP to send other messages, or assign a new message to the button (non persistent))
I wired the two IR leds on GPIO22 using a mosfet, like this:
[image]

Now, after that prototype was done, I discovered the community had gone way beyond this project and the protocol has since been almost fully reverse-engineered (contrary to this project which basically plays back pre-recorded messages, without really understanding their meaning). The best documentation to date probably is the docs folder of https://github.com/jamesw343/PixMob_IR
So if you want to go further than this basic prototype, I suggest you investigate that routes. 
Since then, I've gone overboard and have packed all in a modified IR remote control with a full palette of effects, but that's another story :-)

[image]

[image]

Have fun,

Vicne