danielweidman/pixmob-ir-reverse-engineering #59: PIXMOB TOURMALINE V0.5.0 (RF) [Shakira Americas]

tribakzero, 2025-09-02:
Got some of these, of which I can't find any reference here. It's definitely RF and I'm only recently reading on how to get it tested using a Flipper Zero. The are similar in shape and band style to Karol G's but those are IR, so PIXMOB are really trying to put a bunch of models and styles as possible. These have the technical model: W-4-RGB-009A-AA-1-2 are could possibly be manufactured by may 2025 if I'm reading the Lot properly: VN-2505-4946.

I love how they crafted the antenna across the lower part of the flexible PCB. Happy to try and test whatever you think could help me get these back alive.

[image]

[image]

[image]

[image]

poolster, 2026-05-25:
I got my hands on a few of these after a concert and I want to give it a crack to try and hack them, since I'm a LoRa engineer and they seem to be working in that band. Yours has a 26.1322 MHz crystal which probably relates to a working frequency in the 915 MHz band. Mine has a crystal of 24.8117 MHz and is the European version, problably working somewhere in the 868 MHz band. 
The IC U3 on the right side is most likely a simple OOK (On-Off-Keying) receiver. It could be the HopeRF CMT2210LH or something comparable. Some characteristics that match with the CMT2210LH: it comes in a SOP8 package (the IC in the device seems to be SOP8 or SOIC8). the crystal seems to be connected to pin 8. GND seems to be connected to pin 1. The data output seems to be pin 5, and the RF-input part seems to be connected to the lower side of the IC's, not well visible, but looks like pin 2, 3, or 4. The SMD coils and capacitors located at the feed point of the antenna also seem to match the reference design listed in the CMT2210LH datasheet.
I hooked up the device to my Otii to see the power usage profile and to my logic analyzer to see the data output of the RF-receiver.
After powerup it uses 10 mA for exactly 2 minutes, then it goes in sleep mode. In sleep mode it uses only a few nanoAmps, and in sleep mode there is also a heartbeat signal every 140 ms. 
Every (exactly) 30 seconds, the device comes out of sleep mode to listen for an RF command. The circuit switches on for 250 ms, uses 12 mA during that timeframe indicating the receiver is on, and if no RF-package is received, it goes back to sleep. 
Whenever the receiver is on with no RF-signal present, the data output just seems to transmit continuous random data; looks like demodulated noise. 
I'll be setting up an RF-transmitter to transmit a square-wave modulated carrier, so I can vary the center frequency until I see that square wave on the data output of that RF-receiver, then I'll know the frequency. Then next comes cracking the protocol!

macolmenerori, 2026-11-08:
Hi everyone, I just created [an issue](https://github.com/weeknds/pixmob-wave2-flipper-zero/issues/2) related to this band on [another project](https://github.com/weeknds/pixmob-wave2-flipper-zero). I added support for this wristband model on that project. The info on this issue was very useful and thanks to it I could make it work.